supperthomas/bluetoothlover_doc #87: [OS] docker 使用

supperthomas, 2021-06-24:


supperthomas, 2021-08-29:
嵌入式有哪些应用场景。

xupenghu, 2021-09-12:
Dockerfile的使用 https://blog.csdn.net/u014421520/article/details/120248498

xupenghu, 2021-09-12:
Docker常用命令 https://blog.csdn.net/u014421520/article/details/120248405?spm=1001.2014.3001.5502

xupenghu, 2021-09-12:
Docker容器数据卷 https://blog.csdn.net/u014421520/article/details/120248472?spm=1001.2014.3001.5502

xupenghu, 2021-09-12:
Dokce镜像的理解 https://blog.csdn.net/u014421520/article/details/120248419?spm=1001.2014.3001.5502

supperthomas, 2021-09-28:
@supperthomas 研究一下docker 如何使用

supperthomas, 2021-10-05:
https://mp.weixin.qq.com/s/ueD5htayKhCnzQ5C4ngJqQ

supperthomas, 2021-11-04:
官方docker教程：https://training.play-with-docker.com/

supperthomas, 2021-11-07:
目前掌握dockerfile。https://github.com/supperthomas/Dockerfiles

xupenghu, 2021-11-16:
# 发布自己的镜像到docker hub
1. 首先需要给自己的镜像打一个tag，
```
# docker tag esp32_adf_build:0.1 hubertxxu/esp32_adf_build:latest
```
2. 然后push到docker hub
```
docker push hubertxxu/esp32_adf_build:latest
```
> 如果直接push会报错，因为自己做的docker镜像是没有加docker hub的用户名的。

supperthomas, 2021-11-16:
docker 从入门到精通
https://vuepress.mirror.docker-practice.com/

hyhkjiy, 2021-11-17:
Dockerfile最佳实践 https://vuepress.mirror.docker-practice.com/appendix/best_practices/

supperthomas, 2021-11-21:
数据卷使用：
docker run -it -v 宿主机绝对路径目录: /容器内目录 镜像名

supperthomas, 2021-11-21:
docker run -it -w /root/rtthread -v /f/04_git_repo/GITEE/test/rt-thread_github:/root/rtthread lt6210925/rtthread

supperthomas, 2021-11-22:
docker跑起来还是挺耗资源的

supperthomas, 2022-02-10:
docker 镜像   https://help.aliyun.com/document_detail/60750.html

supperthomas, 2022-03-08:
查找docker hub里面的image
```
docker search supperthomas
```

查看某个dockerhub image的所有tag
直接去对应的tags里面网页里查看

https://registry.hub.docker.com/r/supperthomas/zephyr/tags

supperthomas, 2022-03-09:
docker 如何把USB连到电脑上：

```
--device=/dev/bus/usb
```

supperthomas, 2022-04-16:
https://mp.weixin.qq.com/s/DP5kTaVnIyj5HnGjpKIEVg

supperthomas, 2024-11-01:
ubuntu下面docker安装https://docs.docker.com/engine/install/ubuntu/

supperthomas, 2024-11-01:
https://blog.csdn.net/qq_35628900/article/details/141471981
docker 安装

supperthomas, 2024-12-30:
https://mp.weixin.qq.com/s/Enz5pcT8C_vkD1voz2ic0Q