supperthomas/bluetoothlover_doc #103: [CYC][TASK] nordic 52840上适配USB

chenyingchun0312, 2021-06-27:


chenyingchun0312, 2021-07-26:
切片任务：
1. 了解USB相关资料文档位置，文档大概讲的内容、作用，知道遇到协议不清楚的问题时，翻阅哪一个文档
2. 结合各种硬件设备-鼠标/串口/U盘/AI加速棒等USB抓包，并结合协议文档，分析理解协议
3. 在STM32L开发板上，在rtthread环境中，跑通USB device(CDC类)功能，理解rtthread的usb协议框架
4. 在NRF52840平台上，跑通USB device（CDC类）
5. 在整个学习过程中，总结输出文档，文档应该包含：USB基础知识，USB协议，USB应用按理

chenyingchun0312, 2021-07-26:
讲USB还不错的博文： https://blog.csdn.net/weiaipan1314/category_10697813.html

chenyingchun0312, 2021-07-26:
STM32 USB HOST驱动4G模块参考 https://club.rt-thread.org/ask/article/2750.html

chenyingchun0312, 2021-07-27:
开源usb协议栈，支持usb device 和 usb host，支持nrf5x

https://github.com/hathach/tinyusb

chenyingchun0312, 2021-07-28:
一遍分析目前比较主流USB协议栈的文章，涉及到的很多开源协议栈，包括了rtthread的usb协议栈，还有思维导图，有助于理解rttrehad的usb协议栈

https://forum.butian.net/share/169

chenyingchun0312, 2021-07-28:
rtthread usb协议栈思维导图

[image]

[image]

chenyingchun0312, 2021-08-02:
rt-thread 比较相关的一些博文：
https://blog.csdn.net/rtthreadiotos/category_10584352.html

chenyingchun0312, 2021-08-11:
目前52840跑通了官方的USB Device CDC Demo，官方的usb stack还没有细看，接下来：
- 基本看懂nordic官方的 usb stack
- 捋顺stm32的 rt-thread的usb stack
- 移植，适配rt-thread的usb-stack

chenyingchun0312, 2021-08-21:
nrfx替换nrf_drv
```
		nrfx_clock_init(clk_event_handler);
    nrfx_clock_enable();

//    ret = nrf_drv_clock_init();
//    APP_ERROR_CHECK(ret);
    
		nrfx_clock_lfclk_start();
//    nrf_drv_clock_lfclk_request(NULL);

//    while(!nrf_drv_clock_lfclk_is_running())
		while(!nrfx_clock_lfclk_is_running())
    {
        /* Just waiting */
    }
```

chenyingchun0312, 2021-08-21:
从nrf_drv驱动（nordic老版本驱动）迁移到新版本驱动nrfx驱动，可以参考链接

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_nrf5_v17.0.0%2Fnrfx_migration_user_guide.html&anchor=getting_started_replace_nrf_drv_options_migrate

chenyingchun0312, 2021-08-22:
驱动接口对接，基本OK，测试USB中断可以正常产生，但是枚举时IN DATA，Device并没有发送成功
[image]

```
// 发送IN data失败
NRFX_USBD_TRANSFER_IN(transfer, buffer, size, 0);
nrfx_err_t err = nrfx_usbd_ep_transfer(address, &transfer);
```

chenyingchun0312, 2021-08-29:
EP0 DMA写中断没有产生，后续参考 tinyusb中的nrf52840的驱动查找原因

supperthomas, 2021-08-29:
@chenyingchun0312  提交文档，提交代码到PR.

supperthomas, 2021-08-29:
[full.zip](https://github.com/supperthomas/supperthomas_doc/files/7072416/full.zip)

chenyingchun0312, 2021-08-29:
试试软件usb抓包器  [usblyzer](http://www.usblyzer.com/)

chenyingchun0312, 2021-09-12:
计划：
1. 52840 的USB协议栈，先试用RTT的USB协议栈跑一下，如果比较困难，再切换到tinyusb中跑起来
2. 考虑将tinyusb做成RTT的软件包

chenyingchun0312, 2021-10-17:
https://github.com/supperthomas/supperthomas_doc/issues/169

closed 2021-11-01 by chenyingchun0312

Jackistang, 2021-12-19:
不错的 USB 入门系列文章：[图解 USB](https://mp.weixin.qq.com/mp/appmsgalbum?action=getalbum&__biz=MzU2MDgyNzgyMw==&scene=24&album_id=1458828634189660164&count=3#wechat_redirect)

chenyingchun0312, 2022-05-18:
[USB中文网站](http://www.usbzh.com/)，资料比较全面