supperthomas/bluetoothlover_doc #122: [hex]HEX文件编辑工具

supperthomas, 2021-07-14:


chenyingchun0312, 2021-07-14:
说说这是打算干嘛呢？具体些？ultraedit或者winhex能满足你的需求么

lt6210925, 2021-07-14:
> 说说这是打算干嘛呢？具体些？ultraedit或者winhex能满足你的需求么

不是的，intel hex文件。烧入固件应用的。编辑hex文件的tool

supperthomas, 2021-07-14:
> 
> 
> 说说这是打算干嘛呢？具体些？ultraedit或者winhex能满足你的需求么

这边指的是intel的firmware 的hex格式，搞一个可以编辑和修改合并的工具

lt6210925, 2021-07-14:
https://mp.weixin.qq.com/s/d6UDXs6FwgE_XfEI8ECW_A

supperthomas, 2021-07-14:
https://mp.weixin.qq.com/s/d6UDXs6FwgE_XfEI8ECW_A

supperthomas, 2021-07-14:
https://baike.baidu.com/item/hex%E6%96%87%E4%BB%B6%E6%A0%BC%E5%BC%8F/1025048?fr=aladdin

supperthomas, 2021-07-14:
:1000000000040020810A000015070000610A0000BA

第1个字节10，表示该行具有0x10个数据，即16个字节的数据；

第2、3个字节C000，表示该行的起始地址为0xC000；

第4个字节00，表示该行记录的是数据；

第5-20个字节，表示的是有效数据；

第LENGTH-1个字节BA，表示前面数据的校验数据，校验方法：0x100-前面字节累加和；

00 – 数据记录    => 真实数据
01 – 文件结束记录   => 文件结束
02 – 扩展段地址记录  
04 – 扩展线性地址记录   =>一般在开头

细心的同学可能发现了，上面的地址都是两个字节，范围从0x000-0xFFFF，如果地址是0x17FFFF该怎么办呢？这就要用到扩展字段了，举例如下：
[image]

 第一行中，第一个字节为0x02，表示只有两个字节的数据，而扩展段的标识符为0x04表示后面的数据0x0800为扩展线性地址，基地址的计算方法为：

(0x0800<<16)=0x08000000，在0x04标识段出现之前，下面的数据都是这个基地址。

第二行的地址是0x0000，那么实际地址应是0x08000000+0x0000=0x08000000；

第二行的地址是0x0010，那么实际地址应是0x08000000+0x0010=0x08000010；

使用Notepad++工具，可以根据颜色的不同来确认校验数据是否正确，如果校验数据的颜色不是绿色，则表示校验结果是错的。

supperthomas, 2021-07-14:
https://en.wikipedia.org/wiki/Intel_HEX

supperthomas, 2021-07-14:
https://mp.weixin.qq.com/s/xvDBFn5SSPyf0R0r8moV5w

supperthomas, 2021-07-14:
:04000005080001AD41
<html><body>

05 | 表示开始线性地址
-- | --

</body>
</html>
080001AD 这个地址似乎是main函数的地址
    __main                                   0x080001ad   Thumb Code     8  __main.o(!!!main)

closed 2021-07-16 by supperthomas

supperthomas, 2021-08-10:
http://srecord.sourceforge.net/

supperthomas, 2022-06-11:
使用Jlink来合并烧写文件，就是这么简单

https://baijiahao.baidu.com/s?id=1717088871428920797&wfr=spider&for=pc&searchword=jlink%20%E5%90%88%E5%B9%B6bin

supperthomas, 2022-06-11:
bintohex
https://www.keil.com/download/docs/113.asp

supperthomas, 2022-08-03:
https://github.com/WerWolv/ImHex

supperthomas, 2022-08-19:
IAR如何下载elf文件调试
http://www.bmrtech.com/upload/tech_pdf/000/000/001/5eba41c0d70db.pdf

supperthomas, 2022-10-21:
hex合并工具
https://gitee.com/safe-already-exists/merge-hex

supperthomas, 2022-11-14:
如何用jlink进行烧入