supperthomas/bluetoothlover_doc #124: [USB分析工具]  USBlyer

supperthomas, 2021-07-19:
USBlyer 破解版

supperthomas, 2022-03-04:
https://www.armbbs.cn/forum.php?mod=viewthread&tid=104363

supperthomas, 2022-03-04:
https://www.armbbs.cn/forum.php?mod=viewthread&tid=110118&extra=page%3D1

supperthomas, 2022-03-04:
https://github.com/greatscottgadgets/luna

supperthomas, 2022-03-04:
https://www.armbbs.cn/forum.php?mod=viewthread&tid=95884&extra=page%3D4

WIRESHARK 抓USB

supperthomas, 2022-03-20:
USB lyzer 安装之后，电脑USB口不能用了。这个是个问题，只能虚拟机里面用

supperthomas, 2022-08-09:
3、USBDeview - USB驱动查看和卸载软件
USBDeview 是一个小的实用程序，列出了当前连接到 PC 的所有 USB 设备，以及以前使用的所有USB设备。USBDeview还允许卸载以前使用的 USB 设备，断开当前连接到 PC 的 USB 设备，以及禁用和启用 USB 设备。

工具地址：http://www.nirsoft.net/utils/usb_devices_view.html

supperthomas, 2022-12-01:
USB-HID-TEST 是 jakey 大神用 python2.7、Tkinter、pywinusb(pyusb-linux) 模块开发的 USB 调试工具。  

> 工具主页： https://gitee.com/jakey.chen/USB-HID-TEST

chenyingchun0312, 2023-04-23:
【usb协议基础视频教程-哔哩哔哩】 https://b23.tv/ytsZFsw

supperthomas, 2023-08-24:
https://www.usbzh.com/index.html