supperthomas/bluetoothlover_doc #152: H750 bootloader

supperthomas, 2021-08-16:
https://github.com/RT-Thread/rt-thread/pull/4503

supperthomas, 2021-08-17:
https://github.com/u-boot/u-boot/blob/51aef405550e603ff702c034f0e2cd0f15bdf2bb/cmd/load.c

supperthomas, 2021-08-17:
考虑支持的传输协议：
Xmodem
Ymodem
Zmodem
支持的文件格式：
hex
srec
raw_binary

xupenghu, 2021-08-17:
- 支持差分升级
- 支持加密
- 有配套的上位机

supperthomas, 2021-08-17:
1. 上位机可以负责解析文件格式转换成数组或者hex发送下去。

Jackistang, 2021-08-17:
[rt-fota](https://gitee.com/spunky_973/rt-fota) - 基于 nano 的 bootloader，支持 rbl 文件格式下载。

supperthomas, 2021-08-18:
#122

supperthomas, 2021-08-18:
https://www.keil.com/download/docs/4.asp

chenyingchun0312, 2021-08-24:
一个标星比较多的嵌入式boot
https://www.github.com/wolfSSL/wolfBoot/tree/master/

supperthomas, 2021-09-08:
https://mp.weixin.qq.com/s/T4l_XGl81s9IsQoiUyJ3kQ

supperthomas, 2021-09-15:
三段

supperthomas, 2021-09-16:
https://mp.weixin.qq.com/s/ZrbkleRny3xopMn8PcBQuw

supperthomas, 2021-09-18:
https://mp.weixin.qq.com/s/A-LUF0Cauz1D-ZIdLATmGQ

supperthomas, 2021-09-30:
[MCU固件升级的几种Flash划分方式

](https://mp.weixin.qq.com/s/Vw5500WigTRVTmvRB7ODsA)

supperthomas, 2021-11-03:
【IAR】 如何从APP进行调试：
debug的时候修改如下参数
--drv_vector_table_base=0x0

supperthomas, 2021-11-18:
https://mp.weixin.qq.com/s/W4VFc90NVtX-wrJ93LhnaQ

supperthomas, 2021-11-18:
代码放到RAM中执行

supperthomas, 2021-12-20:
代码VTOR偏移有对应的要求。查看spec

supperthomas, 2021-12-23:
https://mp.weixin.qq.com/s/INpgYXe51xognvWJs59UIw

Jackistang, 2022-01-09:
[From Zero to main()](https://interrupt.memfault.com/blog/zero-to-main-1) 系列

supperthomas, 2022-04-13:
https://mp.weixin.qq.com/s/G-OxnX-0SippkTWNw_3iWA