supperthomas/bluetoothlover_doc #154: [bsp][IIC]研究一下IIC

supperthomas, 2021-08-19:
nordic 的IIC 优化一下

supperthomas, 2021-09-06:
https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/device/i2c/i2c

supperthomas, 2021-09-06:
https://en.wikipedia.org/wiki/I%C2%B2C

supperthomas, 2021-09-06:
模式[10]	最大
速度	最大
电容	驾驶	方向
标准模式 (Sm)	100 kbit/s	400 pF	开漏*	双向
快速模式 (Fm)	400 kbit/s	400 pF	开漏*	双向
快速模式加 (Fm+)	1 兆比特/秒	550 pF	开漏*	双向
高速模式 (Hs)	1.7 兆比特/秒	400 pF	开漏*	双向
高速模式 (Hs)	3.4 兆比特/秒	100 pF	开漏*	双向
超快模式 (UFm)	5 兆比特/秒	?	推拉	单向

supperthomas, 2021-09-06:
1. 所有数据主机发起，从机不会主动发起
2. 7bit地址的寻址是127个地址，如果总线上的地址相同的话，必须要换掉，一根总线上的地址不能有相同的
3.

supperthomas, 2021-09-06:
[嵌入式开发中i2c协议是怎么玩的？](https://mp.weixin.qq.com/s/EMOpvdy6JV_pvlV_KV2Gmg)

supperthomas, 2021-09-06:
速度：
1. 普通模式： 100kHz
2. 快速模式： 400kHz
3. 高速模式： 3.4MHz

supperthomas, 2021-09-06:
[取代I2C和SPI的I3C接口来了，更快速度，更低功耗，兼容I2C](https://mp.weixin.qq.com/s/DihWlJ9aYhpOl3H11hH6Aw)

supperthomas, 2021-09-06:
RT-THREAD 软件包：
[i2c-tools](https://github.com/wuhanstudio/rt-i2c-tools)

supperthomas, 2021-09-09:
主机读数据的时候，如果出现数据都是0xFF，很有可能是从机没回数据

chenyingchun0312, 2021-10-02:
#207

supperthomas, 2023-08-21:
https://mp.weixin.qq.com/s/oVdrIVVRCpfLX_BtNZFk3Q

I3C和I2C之间的差异

supperthomas, 2024-01-05:
https://mp.weixin.qq.com/s/2MIMbp4vl990Et-d20njBg

IIC注意点