supperthomas/bluetoothlover_doc #157: [rt-thread] 自动化format.py脚本

supperthomas, 2021-08-24:
将格式化作为自动化的动作，放在git commit 之前。

chenyingchun0312, 2021-08-29:
git commit hook

chenyingchun0312, 2021-09-14:
开发format GUI程序-支持[formatting](https://github.com/mysterywolf/formatting)脚本的一些功能：

1. 将源文件编码统一为UTF-8
2. 将TAB键替换为4空格
3. 将每行末尾多余的空格删除，并统一换行符为'\n'

supperthomas, 2022-05-13:
代码分析
analytics.google.com

supperthomas, 2023-04-06:
基于cppcheck的静态测试编码规则检查

https://mp.weixin.qq.com/s/jpkeb8egdWn2nK0mh_eBEQ

supperthomas, 2023-04-13:
```
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# cppcheck可以使用cmake文件生成的compile_commands.json，其通过命令行参数指定
# 由于实测compile_commands.json是在CMAKE_CACHEFILE_DIR目录下生成，由此有下面的代码
find_program(CPPCHECK_PROC NAMES cppcheck)
if (CPPCHECK_PROC)
  list(
    APPEND CPPCHECK_CONFIG
    "--enable=warning,performance,information"
    "--project=${CMAKE_BINARY_DIR}/compile_commands.json"
    "--suppress=missingIncludeSystem"
    "--language=c++"
    "--std=c++11"
    "--bug-hunting"
    "--max-ctu-depth=10"
    "-j 4"
    "--output-file=${CMAKE_BINARY_DIR}/cppcheck_report.txt"
  )

  add_custom_target(cppcheck_code_check)

  add_custom_command(TARGET cppcheck_code_check
    COMMAND ${CPPCHECK_PROC} ${CPPCHECK_CONFIG})
endif ()

```

supperthomas, 2023-04-16:
cppcheck --project=compile_commands.json

supperthomas, 2023-04-16:
```

CPUS=$(cat /proc/cpuinfo | grep "processor" | sort | uniq | wc -l) 
```
如何查询CPU个数。
这句话是一个 Linux 命令行脚本中的一部分，用于计算系统中运行着的 CPU 核心数量。下面是每个部分的解释:

cat /proc/cpuinfo:此命令将读取/proc/cpuinfo文件,该文件包含有关系统CPU的信息,例如处理器数量、核心数量、缓存大小等。

grep "processor":此命令将在/proc/cpuinfo 文件中查找“processor”字符串，这意味着仅查找包含“processor”字符串的行。

sort:此命令将匹配的行按字母顺序排序。

uniq:此命令将排序后的行中重复的行删除，只保留一个副本。

wc -l:此命令将计算文件行数，即将 grep、sort 和 uniq 命令返回的行数计算在内。

因此，整个命令将读取/proc/cpuinfo 文件，查找包含“processor”字符串的行，将这些行按字母顺序排序，删除重复的行，并计算文件的行数，以便确定系统中运行着的 CPU 核心数量。

```
cat /proc/cpuinfo  
Architecture:      x86_64  
CPU MHz:          2999.999  
Clock cycles:        29999999  
Maximuminstructions executed: 4095  
MaximumCycles per instruction: 8  
CPU name:         Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz  
Number of processors: 2  
Product ID:         0x1a2a3243  
Processors active: 1  
 Threads active:    2  
```

supperthomas, 2023-04-16:
python3 -m pip install scons
python3 -m SCons -v

+env.Tool('compilation_db')
+env.CompilationDatabase()

sconstruct

supperthomas, 2023-04-16:
pip install scons-compiledb

supperthomas, 2023-04-24:
release 生成exe
https://github.com/jiro4989/sayhissatsuwaza/blob/master/.github/workflows/release.yml

supperthomas, 2023-04-24:
当使用 asset_content_type 输入来指定要上传的 asset 的 MIME 类型时，你可以使用以下参数来指定不同的 asset 类型:

    application/gzip: 用于上传压缩文件，如 GZIP 压缩的 PDF 文件或 ZIP 文件。
    application/octet-stream: 用于上传未知文件类型或未压缩文件。
    application/vnd.github.v3+json: 用于上传 GitHub 版本控制中的 JSON 文件。
    image/png: 用于上传 PNG 格式的图像文件。
    image/jpg: 用于上传 JPEG 格式的图像文件。
    audio/mpeg: 用于上传 .mpeg 格式的音频文件。
    audio/wav: 用于上传 .wav 格式的音频文件。
    video/mp4: 用于上传 MP4 格式的视频文件。
    video/webm: 用于上传 WebM 格式的视频文件。

请注意，这些参数只是一部分，还有其他参数可用于指定不同类型的 asset。具体来说，你可以使用以下参数来指定不同类型的 asset:

    asset_name: 指定要上传的 asset 的名称。
    asset_path: 指定要上传的 asset 的路径。
    upload_url: 指定要使用的上传 URL。
    asset_content_type: 指定要上传的 asset 的 MIME 类型。
    asset_size: 指定要上传的 asset 的大小。
    asset_md5: 指定要上传的 asset 的 md5 哈希值。
    asset_sha1: 指定要上传的 asset 的 sha1 哈希值。
    asset_download_url: 指定要使用的下载 URL。

supperthomas, 2023-04-24:
https://github.com/supperthomas/formatting/releases

chenyingchun0312, 2023-04-25:
> release 生成exe https://github.com/jiro4989/sayhissatsuwaza/blob/master/.github/workflows/release.yml

这是啥意思？

supperthomas, 2023-04-25:
> > release 生成exe https://github.com/jiro4989/sayhissatsuwaza/blob/master/.github/workflows/release.yml
> 
> 这是啥意思？下次给我们介绍介绍呗

好的，下次介绍

supperthomas, 2023-04-25:
标题：如何在github上发布自己软件的可执行文件

supperthomas, 2023-04-25:
https://github.com/supperthomas/formatting

supperthomas, 2023-04-25:
可以创建一个包含以下步骤的GitHub Action：

1. 安装所需依赖：在步骤中，你需要安装 Python 和其他用于生成可执行文件的依赖，例如构建工具和代码库。

2. 生成可执行文件: 执行 python release.py 命令生成可执行文件，并将其构建在 Mac、Windows 和 Linux 上。

3. 创建一个 release: 通过使用官方的 actions/create-release action 显示一个 release 表单，用于填写要发布的版本号和其他信息。

4. 以文件上传形式上传生成的文件: 使用官方的 actions/upload-artifact action，分别将 Mac、Windows 和 Linux 生成的可执行文件上传为 asset。

以下是 GitHub Action 工作流程的参考代码：

name: Release

on:
  push:
    branches:
      - master

jobs:
  build:
    runs-on: ${{ matrix.os }}

    strategy:
      matrix:
        os: [macos-latest, windows-latest, ubuntu-latest]

    steps:
    - name: Checkout code
      uses: actions/checkout@v2

    - name: Set up Python ${{ matrix.python-version }}
      uses: actions/setup-python@v2
      with:
        python-version: ${{ matrix.python-version }}
        architecture: ${{ matrix.arch }}

    - name: Install dependencies
      run: |
        pip install -r requirements.txt
        pip install pyinstaller
      
    - name: Build Executables
      run: |
        pyinstaller -F release.py

    - name: Create Release
      uses: actions/create-release@v1
      env:
        GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}
      with:
        tag_name: ${{ github.ref }}
        release_name: ${{ github.ref }}
        body: |
          Release ${{ github.ref }}
        draft: false
        prerelease: false

    - name: Upload Executables
      uses: actions/upload-artifact@v1
      with:
        name: Executable
        path: dist/release

上述代码适用于在 Mac、Windows 和 Linux 上使用 Python 3 的情况。如果你需要使用 Python 2 或其他依赖，请相应更改代码。同时，需要将 requirements.txt 替换为你的项目所需的依赖列表。执行完毕后，你就可以通过 releases 页面发布可执行文件了。

Join ChatGPT | Chart | Buy | 47.0s

supperthomas, 2023-04-25:
name: Generate Release

on:
  push:
    branches:
      - master
  workflow_dispatch:

jobs:
  build:
    runs-on: ${{ matrix.os }}
    strategy:
      matrix:
        os: [macos-latest, ubuntu-latest, windows-latest]
        python-version: [3.6, 3.7, 3.8, 3.9]

    steps:
      - name: Checkout Repository
        uses: actions/checkout@v2

      - name: Set up Python ${{ matrix.python-version }}
        uses: actions/setup-python@v2
        with:
          python-version: ${{ matrix.python-version }}

      - name: Build Executables
        run: |
          python release.py py2exe
          python release.py py2app
          python release.py pyinstaller

      - name: Create Release
        uses: actions/create-release@v1
        env:
          GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}
        with:
          tag_name: ${{ github.ref }}
          release_name: Release ${{ github.ref }}
          body: |
            This is an auto-generated release.
          draft: false
          prerelease: false

      - name: Attach macOS Executables
        uses: actions/upload-artifact@v2.2.0
        with:
          name:-app-bundle
          path: ./dist/app/mac/*

      - name: Attach Windows Executables
        uses: actions/upload-artifact@v2.2.0
        with:
          name: exe-files
          path: |
            ./dist/app/win32
            ./dist/app/win64

      - name: Attach Linux Executables
        uses: actions/upload-artifact@v2.2.0
        with:
          name: linux-binaries
          path: ./dist/app/linux/*