supperthomas/bluetoothlover_doc #169: 【rtt usb软件包】尝试把tinyusb或者其他usb stack做成rtt 软件包

chenyingchun0312, 2021-08-29:


chenyingchun0312, 2021-09-29:
https://github.com/RT-Thread/rt-thread/issues/4776

chenyingchun0312, 2021-10-01:
rt-thread 代码中，加入tinyusb以及相关nrf52840 移植文件后，可以USB CDC可以正常枚举

接下来研究下原理，然后思考如何将tinyusb做成rt-thread的软件包

[image]

chenyingchun0312, 2021-10-01:
tinyusb KEIL 工程：

https://github.com/NikitaKhalyavin/tinyusb_stm32f103_test_in_keil

https://github.com/nickfox-taterli/tinyusb_stm32f769disco

chenyingchun0312, 2021-10-01:
测试分支： https://github.com/chenyingchun0312/rt-thread/tree/nrfx_usbd

需要继续优化

chenyingchun0312, 2021-10-02:
[image]

chenyingchun0312, 2021-10-02:
tinyusb使用记录

https://mp.weixin.qq.com/s/obowqT02b866YaeWlQWr2g

chenyingchun0312, 2021-10-05:
https://github.com/hathach/tinyusb/discussions/731

chenyingchun0312, 2021-10-17:
目前STM32适配工作基本完成，接着，我这边再把nfr5x适配好

chenyingchun0312, 2021-11-01:
nrf5x usb，目前的阶段，先这样吧。

已完成tinyusb2rtt部分移植，还需要优化
https://github.com/chenyingchun0312/packages/tree/dev_tinyusb
https://github.com/chenyingchun0312/tinyusb2rtt
https://github.com/chenyingchun0312/rt-thread/tree/tinyusb2rtt

状态
nodric 52840 usbd cdc 设备/audio 设备 枚举正常，menuconfig配置

closed 2021-11-01 by chenyingchun0312

supperthomas, 2021-11-01:
有计划PR到upstream吗？