supperthomas/bluetoothlover_doc #170: [doxygen] 讲述一下如何快速简单的部署和生成doxygen

supperthomas, 2021-08-29:


chenyingchun0312, 2021-08-30:
[ 教你如何制作漂亮且专业的程序文档](https://mp.weixin.qq.com/s/HA352TEBELNJ9pgCTMkWIg)

supperthomas, 2021-08-30:
@chenyingchun0312 
有操作过一遍吗？

chenyingchun0312, 2021-08-30:
还没有，先贴一下

closed 2021-09-14 by chenyingchun0312

supperthomas, 2021-11-25:
https://club.rt-thread.org/ask/article/2853.html

supperthomas, 2022-08-03:
一款常用文档生成工具：Doxygen
https://mp.weixin.qq.com/s/XfaP6DF5G6lgXRJo8XAV2Q

supperthomas, 2022-08-03:
修复了一下RT-TRHEAD的doxygen
欢迎尝试使用一下
`https://github.com/RT-Thread/rt-thread/pull/6217`

supperthomas, 2022-08-04:
参考文档

supperthomas, 2022-08-04:
参考论坛
https://club.rt-thread.org/ask/question/677661523354e786.html

supperthomas, 2022-08-06:
https://club.rt-thread.org/ask/article/bf867386b24e5d7c.html

supperthomas, 2022-08-06:
https://docs.zephyrproject.org/latest/connectivity/bluetooth/api/gatt.html#c.bt_gatt_service_register

Forest-Rain, 2022-08-06:
Graphviz是⼀个图形可视化软件。Doxygen使⽤Graphviz的布局引擎dot⽣成各种图形，例如类的继 承关系图、合作图，头⽂件包含关系图等。 安装最新GraphViz 下载地址 http://www.graphviz .org/download/

supperthomas, 2022-08-06:
Doxygen自动生成函数调用图(CallGraph)与文件包含关系图
https://blog.csdn.net/weixin_43829992/article/details/123399853

supperthomas, 2022-08-06:
> Graphviz是⼀个图形可视化软件。Doxygen使⽤Graphviz的布局引擎dot⽣成各种图形，例如类的继 承关系图、合作图，头⽂件包含关系图等。 安装最新GraphViz 下载地址 http://www.graphviz .org/download/

我试了下，没跑起来q(≧▽≦q)q(≧▽≦q)

supperthomas, 2022-08-06:
https://mp.weixin.qq.com/mp/appmsgalbum?__biz=MzkxNDEwNjQ1Nw==&action=getalbum&album_id=2389158368708984832&subscene=189&scenenote=https%3A%2F%2Fmp.weixin.qq.com%2Fs%3F__biz%3DMzkxNDEwNjQ1Nw%3D%3D%26mid%3D2247485204%26idx%3D1%26sn%3Dc825ed00d075d99bbf6e3586d25afdd7%26chksm%3Dc1723622f605bf345b0cad857e6c2f856fd073791864073c335a3e940af850b27d56e9a429a5%26cur_album_id%3D2389158368708984832%26scene%3D189%26ascene%3D78%26devicetype%3Dandroid-29%26version%3D28001757%26nettype%3DWIFI%26abtest_cookie%3DAAACAA%253D%253D%26lang%3Dzh_CN%26exportkey%3DARY%252By1RpQfC4DaeQpeVEwVU%253D%26pass_ticket%3DOiNVY0B4OE6XPQ3ECsFtFIq8fFOrf9v6mGAXPd5yIVVHsluJpbOc%252BvW5wiu70iIk%26wx_header%3D3&nolastread=1#wechat_redirect

supperthomas, 2022-08-06:
https://mp.weixin.qq.com/s/DDWWLvJUZe7o6tJDawgakA

supperthomas, 2023-01-19:
展示
https://supperthomas.github.io/RTT_doxygen_API

supperthomas, 2023-01-27:
DOXYGEN 命令行

```
Doxygen version 1.8.17
Copyright Dimitri van Heesch 1997-2019

You can use doxygen in a number of ways:

1) Use doxygen to generate a template configuration file:
    doxygen [-s] -g [configName]

2) Use doxygen to update an old configuration file:
    doxygen [-s] -u [configName]

3) Use doxygen to generate documentation using an existing configuration file:
    doxygen [configName]

4) Use doxygen to generate a template file controlling the layout of the
   generated documentation:
    doxygen -l [layoutFileName]

    In case layoutFileName is omitted layoutFileName.xml will be used as filename.
    If - is used for layoutFileName doxygen will write to standard output.

5) Use doxygen to generate a template style sheet file for RTF, HTML or Latex.
    RTF:        doxygen -w rtf styleSheetFile
    HTML:       doxygen -w html headerFile footerFile styleSheetFile [configFile]
    LaTeX:      doxygen -w latex headerFile footerFile styleSheetFile [configFile]

6) Use doxygen to generate a rtf extensions file
    RTF:   doxygen -e rtf extensionsFile

    If - is used for extensionsFile doxygen will write to standard output.

7) Use doxygen to compare the used configuration file with the template configuration file
    doxygen -x [configFile]

8) Use doxygen to show a list of built-in emojis.
    doxygen -f emoji outputFileName

    If - is used for outputFileName doxygen will write to standard output.

If -s is specified the comments of the configuration items in the config file will be omitted.
If configName is omitted 'Doxyfile' will be used as a default.
If - is used for configFile doxygen will write / read the configuration to /from standard output / input.

-v print version string
```