supperthomas/bluetoothlover_doc #179: MATTER

supperthomas, 2021-09-09:
调研一下matter协议栈

supperthomas, 2021-09-09:
https://buildwithmatter.com/

supperthomas, 2021-09-09:
https://github.com/project-chip/connectedhomeip

supperthomas, 2021-09-09:
https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nrfconnect

supperthomas, 2021-10-20:
https://mp.weixin.qq.com/s/agdd0OdIba5QiNtcrFtL4A

supperthomas, 2022-03-01:
https://zhuanlan.zhihu.com/p/456295773

supperthomas, 2022-03-18:
User guide
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md

supperthomas, 2022-03-18:
Esp32 c3
https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32

supperthomas, 2022-03-30:
```
--flash_mode dio --flash_freq 80m --flash_size 4MB
0x0 bootloader/bootloader.bin
0x20000 chip-all-clusters-app.bin
0x8000 partition_table/partition-table.bin
0xf000 ota_data_initial.bin
```

[firmware.zip](https://github.com/supperthomas/supperthomas_doc/files/8376611/firmware.zip)

supperthomas, 2022-03-30:
TBD

closed 2022-03-30 by supperthomas

supperthomas, 2022-06-28:
https://mp.weixin.qq.com/s/7LcnUBkn3VkRnXgNqwVBAw

balanceTWK, 2022-09-03:
https://github.com/project-chip/connectedhomeip

supperthomas, 2022-09-30:
乐鑫matter

supperthomas, 2022-11-03:
https://mp.weixin.qq.com/s/7FChuO3ldS9M5wQErQpONw

supperthomas, 2022-11-04:
https://zhuanlan.zhihu.com/p/580325731?utm_psn=1571875736297857025&utm_id=0

supperthomas, 2022-11-23:
https://mp.weixin.qq.com/s/Ykj50pZvEYr3COUsGLyCpg

supperthomas, 2022-11-25:
matter教程
https://www.zhihu.com/people/niu-pi-tang-23-70/posts

supperthomas, 2023-04-11:
nrf52840 dongle FIRMWARE如何制作
https://www.youtube.com/watch?v=xCyzNzYSG4U

supperthomas, 2023-05-06:
https://github.com/project-chip/connectedhomeip/tree/master/src/platform

supperthomas, 2023-05-31:
https://club.rt-thread.org/ask/article/8c92f58a958feeb1.html
https://club.rt-thread.org/ask/article/ab47d2a6e76dc7ae.html
https://club.rt-thread.org/ask/article/b0846ab64046abde.html