supperthomas/bluetoothlover_doc #187: [积分对接]社区积分对接

supperthomas, 2021-09-18:
1.整理需求
2.整理issue
3.整理SOP
https://mp.weixin.qq.com/s/Dbkfdk-Hi5liQTHsEtb2jw

lt6210925, 2021-09-18:
https://docs.qq.com/sheet/DS1Znd2hVSEVrQ3NR?tab=boq7ae&_t=1631952348261

supperthomas, 2021-09-18:
https://docs.qq.com/sheet/DS1Znd2hVSEVrQ3NR?tab=boq7ae&_t=1631952348261

supperthomas, 2021-09-18:
[RT-Thread贡献活动悬赏列表 (1).xlsx](https://github.com/supperthomas/supperthomas_doc/files/7189492/RT-Thread.1.xlsx)

@chenyingchun0312  请帮忙参考列表，列一下Nordic相关的task，后面我来做补充。明年年后开始，每个月发放一次。10月结算一次。

chenyingchun0312, 2021-09-25:
好的

supperthomas, 2021-11-18:
#新手任务 #灭虫计划 🎉📢最轻松的新手Contributor任务，本周继续！

我们正在推出持续的社群活动——灭虫计划，让大家可以零门槛成为TDengine的Contributor。具体规则：
1. TDengine团队以issue的形式（标签为good first issue）发布任务
2. 感兴趣的同学在issue下回复pick up challenge
3. 在收到assigned it to you的回复后，视为认领成功，就可以开始干活啦～
4. 做完任务后记得将Pull Request提交至develop分支哦～

https://github.com/taosdata/TDengine/issues/8692
https://github.com/taosdata/TDengine/issues/8719
https://github.com/taosdata/TDengine/issues/8720
https://github.com/taosdata/TDengine/issues/8721
https://github.com/taosdata/TDengine/issues/8722

closed 2021-11-18 by supperthomas