supperthomas/bluetoothlover_doc #190: 【bsp/nrf5x】sample & ble sample测试

chenyingchun0312, 2021-09-19:
| 测试项目                                   | 测试平台  | 测试结果 |
| ------------------------------------------ | --------- | -------- |
| nrf5x/nrf52832 默认配置编译下载运行        | MDK & GCC | 正常     |
| nrf5x/nrf52832 带协议栈 ble_app_beacon     | MDK       | 正常     |
| nrf5x/nrf52832 带协议栈 ble_app_blinky     | MDK       | 正常     |
| nrf5x/nrf52832 带协议栈 ble_app_uart       | MDK       | 正常     |
| nrf5x/nrf52832 带协议栈 ble_app_hrs        | MDK       | 正常     |
| nrf5x/nrf52832 带协议栈 ble_app_hids_mouse | MDK       | 正常     |
| nrf5x/nrf52832 带协议栈 ble_app_bas        | MDK       | 正常     |

supperthomas, 2021-09-19:
@chenyingchun0312 
https://github.com/supperthomas/nrf5x_sdk/pull/20
陈工已经帮你合进去了。有机会模拟下其他人，从新下最新的代码测一下。看是否有问题

supperthomas, 2021-09-20:
@chenyingchun0312 陈工这个编译下载可以，
测试的时候，有没有测试蓝牙相关 功能？和手机对测？是否正常？

chenyingchun0312, 2021-09-21:
@supperthomas 只测试了nrf connect可以搜索到蓝牙，具体功能没有测试

supperthomas, 2021-09-28:
https://github.com/supperthomas/nrf5x_sdk/issues/22