supperthomas/bluetoothlover_doc #193: Linux常用命令记录

xupenghu, 2021-09-25:
记录常用Linux命令的用法，不定期更新一下

xupenghu, 2021-09-25:
查看一个长文件可以用more命令，也可以用less命令，less可以上下翻页，也可以搜索 用 `/keywords`

chenyingchun0312, 2021-09-25:
linux 三剑客，有时，非常实用，
grep 搜索
awk 分割行，选择某列
sed  查找替换

应用比如 《精准提取MDK工程和SES Segger embeded studio工程的源文件和头文件》, 一起来围观吧 https://blog.csdn.net/u011581240/article/details/120424309?utm_source=app&app_version=4.14.0&code=app_1562916241&uLinkId=usr1mkqgl919blen

xupenghu, 2021-09-26:
嗯 这是一个长期维护的issue 后面的命令会慢慢增加，内容也慢慢丰富 欢迎扩充

xupenghu, 2021-09-26:
```
 # 更新系统时间
 $date -s "2021-09-26 16:44:30"
# 将时间保存到硬件rtc中,下次复位时间还是正常的
$hwclock -w
 ```

xupenghu, 2021-09-27:
# ls -h # 可以将文件大小以我们常用的单位来标识 而不是字节
```
/mnt/sdcard/faceai/resource # ls -lah ai_face_list_123.csv 
-rwxr-xr-x    1 root     root        3.5M Sep 27 17:56 ai_face_list_123.csv
```

xupenghu, 2021-10-07:
# grep
### 作用
在文件中搜寻字符串匹配的行并输出
-i 不区分大小写
-v 排除指定的字符串
### 用法
```
# grep [指定字符串] [文件]
$ grep mysql /root/install.log
```

supperthomas, 2021-11-03:
安装的时候如何搜索软件包：
`apt-cache search xxxx`

xupenghu, 2022-04-06:
## VMware  ubuntu 虚拟机无法识别共享文件夹
- # vmhgfs-fuse /mnt/hgfs/

xupenghu, 2022-05-31:
# Ubuntu 重新获取ip
```
// 清除ip
# dhclient -r   
// 重新获取ip
# dhclient 
```

supperthomas, 2023-05-31:
neofetch
可以查看系统基本信息

supperthomas, 2025-01-11:
查看当前文件夹大小，从大到小排序
```
du -h --max-depth=1 | sort -hr
```