supperthomas/bluetoothlover_doc #199: [NRFX LOG] 如何在nrfx驱动中，使能LOG打印

chenyingchun0312, 2021-09-26:
[image]

预处理，发现：比如针对nrfx_usbd.c的处理中 NRF_LOG_LEVEL 定义成了0，导致预处理，即忽略了打印
[image]

暂且，如下处理，可以打印出来：
[image]

supperthomas, 2021-09-27:
@chenyingchun0312 【青风带你学蓝牙5.0视频教程（nrf52832开发教程/nrf52840开发教程）-哔哩哔哩】https://b23.tv/dgEeuS

chenyingchun0312, 2021-09-27:
经过分析，如下代码，找到解决问题的方法

[image]

其实就是将如下宏，配置成1，然后设置想要的日志level即可
[image]

closed 2021-09-27 by chenyingchun0312