supperthomas/bluetoothlover_doc #201: 【设计模式】设计模式之美学习

chenyingchun0312, 2021-09-27:
https://time.geekbang.org/column/intro/250

[image]

chenyingchun0312, 2021-09-27:
# RTT 代码学习

## 基类 rt_object

```c
/**
 * Base structure of Kernel object
 */
struct rt_object
{
    char       name[RT_NAME_MAX];                       /**< name of kernel object */
    rt_uint8_t type;                                    /**< type of kernel object */
    rt_uint8_t flag;                                    /**< flag of kernel object */

#ifdef RT_USING_MODULE
    void      *module_id;                               /**< id of application module */
#endif
    rt_list_t  list;                                    /**< list node of kernel object */
};
typedef struct rt_object *rt_object_t;                  /**< Type for kernel objects. */

```

## 子类 rt_device

继承`struct rt_object`

```C
struct rt_device
{
    struct rt_object          parent;                   /**< inherit from rt_object */

    enum rt_device_class_type type;                     /**< device type */
    rt_uint16_t               flag;                     /**< device flag */
    rt_uint16_t               open_flag;                /**< device open flag */

    rt_uint8_t                ref_count;                /**< reference count */
    rt_uint8_t                device_id;                /**< 0 - 255 */

    /* device call back */
    rt_err_t (*rx_indicate)(rt_device_t dev, rt_size_t size);
    rt_err_t (*tx_complete)(rt_device_t dev, void *buffer);

#ifdef RT_USING_DEVICE_OPS
    const struct rt_device_ops *ops;
#else
    /* common device interface */
    rt_err_t  (*init)   (rt_device_t dev);
    rt_err_t  (*open)   (rt_device_t dev, rt_uint16_t oflag);
    rt_err_t  (*close)  (rt_device_t dev);
    rt_size_t (*read)   (rt_device_t dev, rt_off_t pos, void *buffer, rt_size_t size);
    rt_size_t (*write)  (rt_device_t dev, rt_off_t pos, const void *buffer, rt_size_t size);
    rt_err_t  (*control)(rt_device_t dev, int cmd, void *args);
#endif

#if defined(RT_USING_POSIX)
    const struct dfs_file_ops *fops;
    struct rt_wqueue wait_queue;
#endif

    void                     *user_data;                /**< device private data */
};
```

## 子类 rt_device_pin

 继承` struct rt_device`

```C
/* pin device and operations for RT-Thread */
struct rt_device_pin
{
    struct rt_device parent;
    const struct rt_pin_ops *ops;
};
```

## 一些思考

- `struct rt_object` 和 `struct rt_device`可以被实例化吗？也就是单独定义如下函数

  ```C
  struct rt_object object;
  struct rt_device device;
  ```

  > 在struct rt_device_pin中定义的struct rt_device parent;不算，这里只是属于 rt_device_pin 对象的一个成员

- `struct rt_object object`和 `struct rt_device device`不能被单独实例化，那是为什么呢？这两个类，又有什么好处呢

- `struct rt_object object` 可以理解为不需要被实例化的普通的类，作用是仅被继承

- `struct rt_device` 可以理解为**抽象类**，定义了一些属性，和抽象方法（想想抽象类的定义）

  > - 只能被继承，不能被实例化
  > - 即可以包含属性，也能包含方法，包含的方法，即可以是抽象方法，也可以不是抽象方法
  > - 子类，继承了抽象类，必须实现抽象类的所有抽象方法，所谓抽象方法，就是指只声明了函数，未定义函数体

​       抽象类，不能被实例化，只能被子类继承，子类复用了抽象类的部分属性，需要实现抽象类的所有抽象方法（C语言中不是标准的抽象类，因此也可以实现部分抽象方法），那我们一起看看`struct rt_device_pin`这个子类，在创建对象时如何操作的？

```
1. 定义struct rt_device_pin 对象
static struct rt_device_pin _hw_pin;

2. 给_hw_pin对象赋值，这里赋值包括父类rt_device,以及子类特殊的方法struct rt_pin_ops *ops;

3. 调用rt_device_pin_register方法给_hw_pin对象赋值，
```

```C
int rt_device_pin_register(const char *name, const struct rt_pin_ops *ops, void *user_data)
{
    
    //////////////////////////////////////////////////////////////////
    // 给父类rt_device赋值//
    _hw_pin.parent.type         = RT_Device_Class_Miscellaneous;
    _hw_pin.parent.rx_indicate  = RT_NULL;
    _hw_pin.parent.tx_complete  = RT_NULL;

#ifdef RT_USING_DEVICE_OPS
    _hw_pin.parent.ops          = &pin_ops;
#else
    _hw_pin.parent.init         = RT_NULL;
    _hw_pin.parent.open         = RT_NULL;
    _hw_pin.parent.close        = RT_NULL;
    // 父类rt_device方法，一般在驱动框架中定义，通用代码，和硬件相关代码，在子类特有的ops中定义，
    // 驱动框架中的通用device操作代码，会调用子类特有ops方法完成中
    _hw_pin.parent.read         = _pin_read;
    _hw_pin.parent.write        = _pin_write;
    _hw_pin.parent.control      = _pin_control;
#endif
	///////////////////////////////////////////////////////////////////
    
    // 给子类自己的特有方法赋值【注意，子类的特有方法，也可在驱动函数中，调用该注册函数之前赋值部分属性】
    _hw_pin.ops                 = ops;
    _hw_pin.parent.user_data    = user_data;

    // 给父类rt_device以及其父类rt_object赋值
    /* register a character device */
    rt_device_register(&_hw_pin.parent, name, RT_DEVICE_FLAG_RDWR);

    return 0;
}
```

这么一分析，大概也能比较了解RTT驱动框架的设计思想吧，其他继承rt_device的对象，也都是这个套路

xupenghu, 2021-09-28:
有没有讲开源C代码的书籍或者视频？ 想学习一个小的 很受欢迎的C项目

supperthomas, 2021-09-28:
https://drive.weixin.qq.com/s?k=ADoAagedAA4yE1WOcsp

supperthomas, 2021-09-28:
https://drive.weixin.qq.com/s?k=ADoAagedAA4iThC1Lx

supperthomas, 2021-09-28:
[image]

uml oopc嵌入式c语言开发精讲

chenyingchun0312, 2021-09-28:
> 有没有讲开源C代码的书籍或者视频？ 想学习一个小的 很受欢迎的C项目 

rtt linux?  哈哈

chenyingchun0312, 2021-10-16:
[image]

closed 2021-11-01 by chenyingchun0312