supperthomas/bluetoothlover_doc #202: 【rt-thread/systick中断】 RTT systick中断函数处理逻辑

chenyingchun0312, 2021-09-28:
疑问：

[image]

解答：
https://club.rt-thread.org/ask/question/433028.html

[image]

也就是说systick中断中，任务切换代码有两个地方：
- 当前任务的时间片结束时，会主动进行任务调度
- `  /* check timer */
    rt_timer_check();` 判断当前是否有其他更高优先级任务处于就绪态，如果有，则切换任务

xupenghu, 2021-09-28:
- 再接再励分析一下 rt_timer_check(); 看一下任务是如何切换的
- 新建任务的时候就自动创建了任务的线程timer，后面的任务阻塞操作都是根据这个timer来完成的

chenyingchun0312, 2021-09-28:
> - 再接再励分析一下 rt_timer_check(); 看一下任务是如何切换的
> - 新建任务的时候就自动创建了任务的线程timer，后面的任务阻塞操作都是根据这个timer来完成的
> 

好的，我分析好了后，贴在这