supperthomas/bluetoothlover_doc #203: [windows][WSL] 使用windows wsl开发的一些常用技巧记录

xupenghu, 2021-09-28:
- windows wsl 1 默认不支持32位程序的编译 可以参考这篇文章来处理 https://kiprey.github.io/2021/04/i386_WSL64/ 
- 建议直接升级为WSL2，默认支持32位程序

xupenghu, 2021-09-28:
- 我用的是WSL Ubuntu20.04 默认自带gcc 9.3 但是有些工程如果用新的gcc来编译会报错，因为gcc/g++高版本新增了很多语法特性
- 如何切换到低版本的gcc？
- 参考这篇文章 https://blog.csdn.net/u014421520/article/details/119445020

supperthomas, 2021-09-28:
Wsl有哪些优劣势？

xupenghu, 2021-09-28:
> Wsl有哪些优劣势？

优势：
1. 轻量化，不需要安装虚拟机再装系统
2. 快速，可以像使用power shell那样来使用Ubuntu，用windows的vscode开发，终端使用的是vsl 开发效率高
3. 绝大部分的功能和虚拟机Ubuntu是一样的

劣势：
1. 没有界面，只有命令行终端
2. 无法使用NFS server

chenyingchun0312, 2021-09-29:
WSL中的文件，好像，只能放在C盘，另外，我目前安装的是WSL1如何，好升级到WSL2吗？

xupenghu, 2021-09-29:
> WSL中的文件，好像，只能放在C盘，另外，我目前安装的是WSL1如何，好升级到WSL2吗？

参考这篇文章 https://kiprey.github.io/2021/04/i386_WSL64/

chenyingchun0312, 2021-10-03:
[微软官网WSL文档](https://docs.microsoft.com/zh-cn/windows/wsl/)

[WSL2安装使用](https://www.jianshu.com/p/0aa542003b93)

supperthomas, 2021-11-07:
后续继续使用。

supperthomas, 2021-11-15:
https://zhuanlan.zhihu.com/p/147233604

xupenghu, 2021-11-16:
发现一个比较难受的确定，WSL 的 IO操作太慢了，具体体现在编译工程的时候超慢。

xupenghu, 2021-11-16:
还有一个缺点 WSL不支持音频 无法播放声音

chenyingchun0312, 2021-11-17:
> 还有一个缺点 WSL不支持音频 无法播放声音

这个具体是指？

xupenghu, 2021-11-18:
> > 还有一个缺点 WSL不支持音频 无法播放声音
> 
> 这个具体是指？

比如无法通过mplayer播放音频，因为WSL不支持音频驱动，要实现还要转发到windows，比较麻烦

supperthomas, 2021-11-19:
https://www.jianshu.com/p/20ae65d1b7a1

chenyingchun0312, 2022-03-01:
将WSL移出C盘

https://blog.csdn.net/yihuajack/article/details/119915303