supperthomas/bluetoothlover_doc #205: Le audio

supperthomas, 2021-09-29:


supperthomas, 2021-09-30:
https://github.com/zephyrproject-rtos/liblc3codec

chenyingchun0312, 2021-09-30:
我们公司做le audio也在做，可惜一直没能直接参与，所以对le audio 其实也比较不了解，后续肯定是需要学习le audio

supperthomas, 2022-01-01:
https://mp.weixin.qq.com/s/QavJvKNhwWjYv5eVWKlbtA

supperthomas, 2022-01-13:
https://www.bluetooth.com/bluetooth-resources/le-audio-book/?utm_source=marketo&utm_medium=email&utm_campaign=le-audio&utm_term=email&utm_content=leaudio-book-announce-enth&mkt_tok=OTE1LUJSRi0xNDkAAAGB8CLlEloMqkOeOZ3sp2n4bbyupKK2OjDu_Iu6Wrec9aCryjKvTPfAFgGBxtFJ79OYcWl4hPDPX4V921t0OjbsEi2B1SRu3-HbusMCjPCriQ

chenyingchun0312, 2022-06-16:
Nordic LE AUDIO开发版

https://mp.weixin.qq.com/s/fhEOxc6BDRD93bHiF0wwxw

supperthomas, 2024-11-08:
1. LE Audio技术概览：架构与协议
深入探讨LE Audio的技术架构和通信协议。
2. LE Audio的编码技术：LC3与SBC的比较
分析LE Audio的LC3编码与现有的SBC编码的性能对比。
3. LE Audio的同步机制：Auracast与广播音频
详细解释LE Audio的Auracast技术及其在广播音频中的应用。
4. LE Audio的多流音频传输技术
探讨LE Audio如何支持多流音频传输及其技术实现。
5. LE Audio的低功耗设计：电源管理与优化
分析LE Audio的低功耗设计原则和电源管理策略。
6. LE Audio的音频质量控制：延迟与抖动
讨论LE Audio如何控制音频传输中的延迟和抖动，以保证音质。
7. LE Audio的设备发现与配对流程
详细介绍LE Audio设备发现和配对的技术细节。
8. LE Audio的音频流控制：QoS与流量管理
探讨LE Audio如何实现音频流的服务质量(QoS)和流量管理。
9. LE Audio的安全性分析：加密与认证机制
分析LE Audio在数据传输中的加密和认证机制。
10. LE Audio的软件开发：API与框架
介绍LE Audio的软件开发接口(API)和框架。
11. LE Audio的测试与调试：工具与方法
探讨LE Audio开发过程中的测试和调试工具及方法。
12. LE Audio的兼容性测试：跨平台与跨设备
分析LE Audio在不同平台和设备间的兼容性测试策略。
13. LE Audio的硬件要求：芯片与模块
讨论实现LE Audio功能所需的硬件要求和推荐的芯片/模块。
14. LE Audio的软件更新与维护：固件与OTA
探讨LE Audio设备的软件更新和远程固件升级(OTA)。
15. LE Audio的高级应用：空间音频与3D音效
讨论LE Audio在空间音频和3D音效中的应用。
16. LE Audio的性能优化：编码效率与数据压缩
分析LE Audio的编码效率和数据压缩技术。
17. LE Audio的开发者指南：最佳实践与常见问题
提供LE Audio开发的最佳实践和常见问题的解决方案。
18. LE Audio的生态系统构建：合作伙伴与标准制定
探讨如何构建LE Audio的生态系统，包括合作伙伴关系和标准制定。
这些标题将帮助你的博客系列深入探讨LE Audio的技术细节，为读者提供丰富的技术知识和实践指导。

supperthomas, 2024-11-12:
https://mp.weixin.qq.com/s/p_ktcks6EGJ76siRrcazKQ

supperthomas, 2024-11-21:
1. 三星 s24系列，支持le audio中的unicast和auracast，连接qcc3084测试正常
2. 三星 a55，测试结果同三星s24系列
3. 三星 a54，测试支持le audio的unicast，没看到手机auracast选项
4. 红米 k70系列，支持le audio的auracast，配合qcc3084测试无法支持unicast，好像必须配合红米自家的耳机才能测试到le audio的unicast
5. 谷歌 pixel 7a，配合qcc3084测试，le audio的unicast无声

supperthomas, 2024-11-25:
从高通《关于LE Audio的问题解答》看新一代蓝牙音频技术标准对市场的影响
https://mp.weixin.qq.com/s?__biz=MzU5NzE0NjE0MQ==&mid=2247549946&idx=1&sn=fd4f1f8a9d85da109ad3db7f5190f337&chksm=fe55ac88c922259ec62081433393bcea581488f1e95db7be21745f7955d6b46f991279298a76&scene=21#wechat_redirect

supperthomas, 2024-12-02:
le audio 耳机
以下是几款支持LE Audio的耳机推荐：

索尼WF-1000XM5

索尼耳机WF-1000XM5和LinkBuds S已通过软件升级支持LE Audio（LC3编码器），提供更低的传输延迟和功耗，适合游戏使用。这些耳机现在支持广泛的设备兼容性，包括非索尼设备。

松下 Technics AZ40M2

这款真无线主动降噪耳机支持LE Audio与LC3技术，具有高音质和先进的降噪功能。它的轻巧设计（单只耳机4.4g）和长达18小时的总续航时间使其成为旅行或日常使用的理想选择。

索尼INZONE Buds

作为INZONE游戏系列的一部分，INZONE Buds支持LE Audio，专为电竞和游戏优化。它的低延迟连接和专为游戏设计的降噪功能，使其成为电竞爱好者的首选。

这些耳机均支持最新的LE Audio技术，提供更好的音质体验和设备兼容性，适合追求高质量音频和便捷使用体验的用户。

supperthomas, 2024-12-05:
三星galaxy buds3	bes2800	https://item.jd.com/100127183384.html	989	6nm制程，支持LE AUDIO
vivoTWS 4 Hi-Fi	高通第三代S3	https://item.jd.com/100104242270.html#crumb-wrap	480	支持LE AUDIO

supperthomas, 2024-12-05:
https://item.jd.com/100155299370.html   redmibus6pro

supperthomas, 2024-12-05:
https://item.jd.com/100070911747.html INZONE Buds