supperthomas/bluetoothlover_doc #207: 【bsp/nrf5x/】drv_i2c.c 驱动实现存在问题，需要修复

chenyingchun0312, 2021-09-30:
`bsp\nrf5x\libraries\drivers\drv_i2c.c` 是配上，有一些问题，不适配Rt-thread I2C驱动框架标准接口

## 上层应用需求
SHT2x 软件包中代码，重点关注下 `rt_i2c_transfer`调用时参数含义
```
static rt_err_t read_regs(struct rt_i2c_bus_device *bus, rt_uint8_t reg, rt_uint8_t len, rt_uint8_t *buf)
{
    struct rt_i2c_msg msgs[2];

    msgs[0].addr = SHT20_ADDR;
    msgs[0].flags = RT_I2C_WR;
    msgs[0].buf = &reg;
    msgs[0].len = 1;

    msgs[1].addr = SHT20_ADDR;
    msgs[1].flags = RT_I2C_RD;
    msgs[1].buf = buf;
    msgs[1].len = len;

    if (rt_i2c_transfer(bus, msgs, 2) == 2)
    {
        return RT_EOK;
    }
    else
    {
        return -RT_ERROR;
    }
}
```

## 错误点: 

对msgs[] 参数，以及num参数，没有完全理解，msgs传过来的是一个数组，数组元素的个数是num，但是我们驱动中，确将msgs数组大小默认为了1

`bsp\nrf5x\libraries\drivers\drv_i2c.c` 
```
static rt_size_t _master_xfer(struct rt_i2c_bus_device *bus,
                                struct rt_i2c_msg msgs[],
                                rt_uint32_t num)
{
    nrfx_twim_t const * p_instance = &((drv_i2c_cfg_t *)bus->priv)->twi_instance;
    nrfx_err_t ret = NRFX_ERROR_INTERNAL;
    uint32_t no_stop_flag = 0;

    nrfx_twim_xfer_desc_t xfer = NRFX_TWIM_XFER_DESC_TX(msgs->addr,msgs->buf, msgs->len);
    if((msgs->flags & 0x01) == RT_I2C_WR)
    {
        xfer.type = NRFX_TWIM_XFER_TX;
        if((msgs->flags & 0x40) == RT_I2C_NO_READ_ACK)
        {
            no_stop_flag = NRFX_TWIM_FLAG_TX_NO_STOP;
        }
    }
    else if((msgs->flags & 0x01) == RT_I2C_RD)
    {
        xfer.type = NRFX_TWIM_XFER_RX;
    }
    ret = nrfx_twim_xfer(p_instance,&xfer,no_stop_flag);
    return (ret == NRFX_SUCCESS) ? msgs->len : 0;

}
```

## 正确示例，需要循环处理
`bsp\gd32303e-eval\drivers\drv_i2c.c`
```
static rt_size_t gd32_i2c_xfer(struct rt_i2c_bus_device *bus, struct rt_i2c_msg msgs[], rt_uint32_t num)
{
    struct rt_i2c_msg *msg;
    rt_uint32_t i;
    rt_err_t ret = RT_ERROR;

    struct gd32_i2c_bus *gd32_i2c = (struct gd32_i2c_bus *)bus;

    for (i = 0; i < num; i++)
    {
        msg = &msgs[i];

        if (msg->flags & RT_I2C_ADDR_10BIT)
        {
            i2c_mode_addr_config(gd32_i2c->i2c_periph,I2C_I2CMODE_ENABLE,I2C_ADDFORMAT_10BITS,0);
        }
        else
        {
            i2c_mode_addr_config(gd32_i2c->i2c_periph,I2C_I2CMODE_ENABLE,I2C_ADDFORMAT_7BITS,0);
        }
        if (msg->flags & RT_I2C_RD)
        {
            if (gd32_i2c_read(gd32_i2c->i2c_periph, msg->addr, msg->buf, msg->len) != 0)
            {
                LOG_E("i2c bus write failed,i2c bus stop!");
                goto out;
            }
        }
        else
        {
            if (gd32_i2c_write(gd32_i2c->i2c_periph, msg->addr, msg->buf, msg->len) != 0)
            {
                LOG_E("i2c bus write failed,i2c bus stop!");
                goto out;
            }
        }
    }

    ret = i;

out:
    LOG_E("send stop condition\n");

    return ret;
}
```

supperthomas, 2021-10-02:
有Pr吗？

chenyingchun0312, 2021-10-02:
> 有Pr吗？

不方便测试，没有pr，看你在优化nordic iic,可以测试pr下

supperthomas, 2021-10-02:
> > 有Pr吗？
> 
> 不方便测试，没有pr，看你在优化nordic iic,可以测试pr下

Nordic的还没看

supperthomas, 2021-10-02:
你觉得nordic的有问题的话，欢迎pr

chenyingchun0312, 2021-10-03:
> 你觉得nordic的有问题的话，欢迎pr

不知道nordic开发版是否可以测试iic?我这边主要没有测试环境

chenyingchun0312, 2021-11-27:
已经PR: https://github.com/RT-Thread/rt-thread/pull/5309

closed 2021-11-27 by chenyingchun0312