supperthomas/bluetoothlover_doc #211: 锻炼身体打卡

supperthomas, 2021-10-05:


supperthomas, 2021-10-05:
[image]

跑走步3公里，上下楼梯11楼3次

chenyingchun0312, 2021-10-06:
哎哟，我去，涛哥，给力呀

chenyingchun0312, 2021-10-06:
我已经很久没有跑步了，跑不动，一般就走

supperthomas, 2021-10-16:
[image]
2021.10.16    走路

supperthomas, 2021-11-07:
比较难坚持。。。。

supperthomas, 2022-01-05:
[image]
周三肩

supperthomas, 2022-01-09:
周六瑜伽，风驰电掣

chenyingchun0312, 2022-01-14:
套餐价格咋样？多少钱，多长时间

supperthomas, 2022-01-14:
> 套餐价格咋样？多少钱，多长时间

没多少钱吧，估计就一两千一年。不建议办健身卡，我都是已经倒闭两家了。从第一家倒闭之后转到第二家，第二家没去几次又倒闭，现在是第三家。在家买个跑步机吧，我刚买的999昨天跑了5公里
[image]

chenyingchun0312, 2022-01-14:
人太胖，一般的跑步机吃不消我，我们家700多买了一个，我往上一站，不动了，尴尬....

supperthomas, 2022-08-08:
要做无氧运动，有氧运动容易减脂

supperthomas, 2022-08-08:
有氧运动，节奏慢，强度小。持续期间长。比如慢跑
无氧运动，节奏块，强度大，持续时间短

supperthomas, 2022-12-01:
2022/11/30   动感单车1个小时7：00-8：30  火鸟健身

supperthomas, 2022-12-08:
> 2022/12/7
单车20分钟，肩颈理疗50+https://user-images.githubusercontent.com/60349489/206586858-f214b209-210e-48f2-b6a6-356b0e2fce76.mp4

supperthomas, 2022-12-09:
2022/12/08  无绳 跳绳  1000

supperthomas, 2023-04-23:
20230424
20俯卧撑，胸部

supperthomas, 2023-04-25:
20230425
俯卧撑20个

supperthomas, 2023-05-13:
最近有点感冒。

supperthomas, 2023-06-29:
6.29

俯卧撑  20个