supperthomas/bluetoothlover_doc #242: 【2021-11-17】 会议讨论议题

supperthomas, 2021-11-06:
1. docker 与RT-THREAD
2. formatting脚本
3. wio terminal
4. 论坛答题感悟，有苦有甜
5. nimble 更新到最新的代码，和主分支同步。

supperthomas, 2021-11-07:
1. 如何更有效率开会。

supperthomas, 2021-11-07:
【2021-11-21】
@xupenghu 
简单聊下电赛

10每个人介绍，10分钟内，快速过。

会议控制1个小时内。

xupenghu, 2021-11-18:
【2021-11-21】议题
1. 聊一聊ESP32-IDF如何构建自己的软件包和项目，ESP-ADF docker编译环境的使用方法
2. ESP8266 RTOS SDK如何构建自己的软件包和项目
3. 音频总谐波失真的定义和测量方式

chenyingchun0312, 2021-11-19:
【2021-11-21】议题
1. 聊聊最近在读的一本数《C嵌入式编程设计模式》，以及设计模式视频课程
2. 其他待做事项，定时器软件包，nrf52840 usb再试一下

hyhkjiy, 2021-11-19:
【2021-11-21】议题
1. 聊聊nRF5 SDK的观察者模式
  - https://github.com/Zhiyi-Zhang/nordic-sdk-nrf5/blob/82988c22ab48925471b5d90a1fcc83b78d5e262d/components/softdevice/common/nrf_sdh_ble.h#L82
  - https://github.com/Zhiyi-Zhang/nordic-sdk-nrf5/blob/82988c22ab48925471b5d90a1fcc83b78d5e262d/components/softdevice/common/nrf_sdh_ble.c#L265

chenyingchun0312, 2021-12-11:
【2021-12-11】议题
1. 聊一聊经典蓝牙A2DP， 以及LE AUDIO的一些基础知识学习

supperthomas, 2021-12-12:
1.action机器人讨论
2. 下次主持人讨论排序
3. Arduino讨论

chenyingchun0312, 2022-01-07:
2022-1-9 chenyingchun:
分享一下LC3的BQB认证的大概流程

supperthomas, 2022-01-08:
2022-1-9
1.arduino仿真
2.pts测试讨论，后续用nordic试试
3.studio nordic
4.nimble

supperthomas, 2022-01-20:
【2022/01/20】:
1. 聊聊低代码编程
2. 聊聊LLSYNC 简单讲讲架构。

chenyingchun0312, 2022-01-20:
【2022/01/22】:
1. 聊聊经典蓝牙和LE Audio的1-N广播

xupenghu, 2022-01-22:
【2022/01/22】:
1. 想在小组内开一个新的专题：《rt-thread、TencentOS-tiny、freeRTOS三个RTOS设计横向对比》，晚上讨论一下有没有必要，以及开展的方式

supperthomas, 2022-03-11:
【2022/03/11】:

- 内网穿刺frp
- 软件包分类讨论和畅想（是不是应该特殊硬件和应用软件分开）[所有软件包列表](https://github.com/supperthomas/rtthread_software_package_list_show/issues/9)
- ble 仿真模拟这是个比较好玩的东西
- llsync
-

chenyingchun0312, 2022-03-29:
【时间待定】：
- chenyingchun
  - 分享一些IPC的学习经验，常见的IPC软件包，以及一些通用的知识点

supperthomas, 2022-03-30:
> 【时间待定】：
> 
>     * chenyingchun
>       
>       * 分享一些核间IPC的学习经验，常见的IPC软件包，以及一些通用的知识点
>         
>         * 问题：
>           
>           1. 多核对比单核有什么优势？（任务调度、进程管理、内存管理等角度）
>           2. 如何指定代码运行到不同的核心，以及执行的原则是什么？
>           3. RTOS是如何管理多核通讯的

RTOS CPU 性能测试跑分等讨论

supperthomas, 2022-05-03:
2022.5.3
1.公众号赞赏
2.jlink远程调试讨论
3.

xupenghu, 2022-05-03:
## 2022.5.3 会议纪要

supperthomas, 2022-05-31:
## 2022.6.4 会议讨论议题

- [ ] 唐嘉分享一下qemu的ble： 带着下面的问题思考一下。
- [ ] 思考一下会议的目标和如何实现的方法
- [ ] 回顾一下之前议题的方式有哪些不足的地方，如何改进
- [ ] 思考一下之前议题有哪些好的地方值得学习的。

xupenghu, 2022-06-25:
## 2022.6.25 会议讨论议题

- [ ] UCM108E基于rt-thread的分享

## 会议录制链接

supperthomas, 2022-08-03:
- 碎片化学习，关心自己兴奋点，不要去管不懂的东西，就围绕自己的兴趣，只去了解自己懂的部分。碎片化学习
- 整理每个人的issue
- 分享一下LVGL

supperthomas, 2022-08-09:
# 会议流程
1. 在线的小伙伴每人更新一下issue，comment前面两周做了哪些工作，没有时间也可以写没有时间（1-20分钟）
2. 最近两周有什么新闻10分钟
3. 主题分享

supperthomas, 2022-08-22:
# 20220903

内卷

supperthomas, 2022-09-01:
【从瞌睡厌学到自律上瘾❗️让学习比打游戏更爽-哔哩哔哩】 https://b23.tv/dGiH4m7

xiangxistu, 2022-09-03:
Here，20220903

balanceTWK, 2022-09-03:
https://github.com/balanceTWK/armv7m_DWT

supperthomas, 2022-09-04:
【完美主义是人性的弱点-哔哩哔哩】 https://b23.tv/OlUcpdJ

supperthomas, 2022-09-14:
20220917
工匠精神
目标，什么是平衡

supperthomas, 2022-09-21:
20221002

- [ ] action
- [ ] qemu
- [ ] stm32
- [ ] information
- [ ]

supperthomas, 2022-10-06:
BLUENRG GUI

bobwenstudy, 2022-10-22:
20221022

任务进展沟通
讲座Time-《从零到一搭建Kconfig配置系统》+《AutoPts简述》
会议流程讨论

TODO LIST:

bobwenstudy, 2022-10-22:
Part1：(opt)任务进展沟通
Part2：讲座Time（主讲+辅讲）
Part3：茶话会+讲座安排（https://docs.qq.com/sheet/DY1lZbEFwSFlLTmlh）

xupenghu, 2022-11-05:
2022/11/05

## 《BLE mesh介绍》
1. BLE mesh简介
2. BLE mesh pb-adv入网
3. 会后讨论

supperthomas, 2022-11-15:
20221119
- [ ] nordic vscode
- [ ] wiki action & release
- [ ]

supperthomas, 2022-11-25:
如何关闭邮件提醒：
到这个页面，可以关掉不想要的提醒。
[image]

supperthomas, 2022-12-15:
当你做一件事情是被别人逼着做的时候，你会效率变低。
大家可以思考一下活着的意义，活着的价值，完成任务的意义。想不明白，做任何事情，都会失去价值感。
当你觉得做这件事情是有意义的时候，你会恨开心的去做。
当思考意义的时候，首先想想对自己有没有意义，对未来的自己，过去的自己有没有意义，如果有意义，再思考一下对别人有没有意义，有没有帮助。

supperthomas, 2022-12-15:
我们不能光盯着不可控的事情，这样我们很难继续往前成长，比如老板怎么样，公司怎么样。
我们要关注可控的，我们能做的东西，因为只有做了才能成长。

supperthomas, 2023-01-13:
【《中国奇谭》 小妖怪的配音幕后 欢迎来到 妖怪之声-哔哩哔哩国创】https://b23.tv/ep719194

xupenghu, 2023-05-13:
# 20230503

《[合宙Air780e C-SDK开发](https://blog.csdn.net/u014421520/article/details/130660974)》
《[【RDC2022纪念板】RT-Smart D1s上手](https://blog.csdn.net/u014421520/article/details/130491554)》

supperthomas, 2024-12-22:
# 2025-01-05
赵工

supperthomas, 2024-12-22:
# 2025-01-19

getting start up

Li-Dongze, 2024-12-22:
# 2025-01-19
主持人—：李东泽，题目：清风蓝牙入门踩坑记录

supperthomas, 2024-12-22:
# 2025-02-02

李涛