supperthomas/bluetoothlover_doc #259: [github][action] 学习action

supperthomas, 2021-11-30:
https://github.com/supperthomas/esp-idf-ci-action

hyhkjiy, 2021-11-30:
官方资料
---
 - 关于持续集成：https://docs.github.com/cn/actions/automating-builds-and-tests/about-continuous-integration
 - GitHub Actions 快速入门：https://docs.github.com/cn/actions/quickstart

supperthomas, 2021-11-30:
[image]

supperthomas, 2021-11-30:
https://blog.csdn.net/alex_yangchuansheng/article/details/108313625

supperthomas, 2021-11-30:
创建 Docker 容器操作
https://docs.github.com/cn/actions/creating-actions/creating-a-docker-container-action

supperthomas, 2021-12-01:
[image]

supperthomas, 2021-12-01:
ubuntu 原装软件
https://github.com/actions/virtual-environments/blob/main/images/linux/Ubuntu2004-README.md

supperthomas, 2021-12-01:
GitHub Actions 有一些自己的术语。

    workflow （工作流程）：持续集成一次运行的过程，就是一个 workflow。

    job （任务）：一个 workflow 由一个或多个 jobs 构成，含义是一次持续集成的运行，可以完成多个任务。

    step（步骤）：每个 job 由多个 step 构成，一步步完成。

    action （动作）：每个 step 可以依次执行一个或多个命令（action）。

supperthomas, 2021-12-01:
[image]

supperthomas, 2021-12-01:
yml 语法
https://docs.github.com/cn/actions/learn-github-actions/understanding-github-actions

supperthomas, 2021-12-01:
[image]

supperthomas, 2021-12-01:
要看懂action.yml 看下面这个连接比较管用
https://docs.github.com/cn/actions/learn-github-actions/understanding-github-actions

supperthomas, 2021-12-01:
所有的参考
https://github.com/actions/checkout
https://github.com/actions

supperthomas, 2021-12-01:
还有个软件包的问题：
1. menuconfig需要 如何能用命令行选择软件包。这个问题比较难解决？

supperthomas, 2021-12-01:
> 还有个软件包的问题：
> 
>     1. menuconfig需要 如何能用命令行选择软件包。这个问题比较难解决？

哈哈，有办法了。。

hyhkjiy, 2021-12-01:
> > 还有个软件包的问题：
> 
> > 
> 
> >     1. menuconfig需要 如何能用命令行选择软件包。这个问题比较难解决？
> 
> 
> 
> 哈哈，有办法了。。
分享一下吧😁

supperthomas, 2021-12-01:
> > > 还有个软件包的问题：
> > 
> > 
> > > 
> > 
> > 
> > > ```
> > > 1. menuconfig需要 如何能用命令行选择软件包。这个问题比较难解决？
> > > ```
> > 
> > 
> > 哈哈，有办法了。。
> > 分享一下吧😁

周会给大家分享一下

supperthomas, 2021-12-01:
> > > 还有个软件包的问题：
> > 
> > 
> > > 
> > 
> > 
> > > ```
> > > 1. menuconfig需要 如何能用命令行选择软件包。这个问题比较难解决？
> > > ```
> > 
> > 
> > 哈哈，有办法了。。
> > 分享一下吧😁

可以先把.config弄好，然后拷贝覆盖进去

supperthomas, 2022-09-21:
https://club.rt-thread.org/ask/article/9683302d4d44f637.html

supperthomas, 2022-12-01:
Github Action 精华指南
https://zhuanlan.zhihu.com/p/164744104

supperthomas, 2022-12-01:
GitHub Actions 入门教程
https://www.ruanyifeng.com/blog/2019/09/getting-started-with-github-actions.html

supperthomas, 2022-12-04:
https://github.com/xqyjlj/xqyjlj

supperthomas, 2022-12-15:
travis
源码在这里 https://github.com/armink/UtestRunner ，
FlashDB 里也有用到 https://github.com/armink/FlashDB/blob/master/.travis.yml

supperthomas, 2023-01-17:
Intro to CI/CD Part 2: Getting Started with GitHub Actions | Digi-Key Electronics
https://www.youtube.com/watch?v=8pyqbYDYkRs

supperthomas, 2023-01-28:
[image]

github action token权限。

supperthomas, 2023-01-28:
``` 
     - name: Push to GitHub
        uses: EndBug/add-and-commit@v9
        with:
          message: "🎆 docs:  Update new data"
```

supperthomas, 2023-01-28:
```

nohup qemu-system-arm -M vexpress-a9 -smp cpus=2 -kernel rtthread.bin -nographic -sd sd.bin > cmd2.out 2>&1 &

```

supperthomas, 2023-02-07:
github action 缓存机制
https://github.com/a1012112796/rtt_ci_env
https://github.com/a1012112796/rt-thread/blob/zzc/dev/ci_simple/.github/workflows/action.yml
https://cloud.tencent.com/developer/article/1695206

supperthomas, 2023-02-07:
通过rtt_ci_env仓库的ci 来push对应的image到dockerhub的不同tag
然后通过rtthread里面的action来调用对应的image。这个其实耗时比较大。

supperthomas, 2023-02-07:
rt-thread :
apt-get install tool 需要30分钟，
安装gcc需要20分钟。
每个bsp都需要这些时间。

supperthomas, 2023-02-07:
参考机制
https://github.com/zephyrproject-rtos/docker-image/pkgs/container/ci

supperthomas, 2023-02-07:
github 关于package 文档
https://docs.github.com/en/packages/working-with-a-github-packages-registry/working-with-the-container-registry

supperthomas, 2023-02-10:
```
        echo CONFIG_RT_USING_UTESTCASES=y >> $TEST_BSP_ROOT/.config
        cat examples/utest/configs/$TEST_CONFIG_FILE >> $TEST_BSP_ROOT/.config
```

supperthomas, 2023-05-11:
$(nproc)

supperthomas, 2023-05-11:
continue-on-error:

supperthomas, 2023-05-11:
ls bsp/$bsp/Kconfig && scons -C bsp/$bsp --pyconfig-silent
            pushd bsp/$bsp && pkgs --update && popd

supperthomas, 2023-05-19:
https://github.com/xpack-dev-tools/qemu-arm-xpack/blob/325ed77bbefdad7ad1aa2732cac6456a24321da9/.github/workflows/build-all.yml#LL26C2-L26C2

supperthomas, 2023-05-19:
https://docs.github.com/en/actions/using-workflows/workflow-commands-for-github-actions#grouping-log-lines

supperthomas, 2023-06-02:
https://github.com/nektos/act
action 本地化运行

supperthomas, 2024-08-26:
做个package 没有toolchain