supperthomas/bluetoothlover_doc #266: [l496][alios things][developkit] 阿里开发板资料收集

supperthomas, 2021-12-05:
整理一些传感器型号和一些外设的型号：

- [ ] 加密芯片： MTF008
[加密 Security Gemalto-MTF008.pdf](https://github.com/supperthomas/supperthomas_doc/files/7656225/Security.Gemalto-MTF008.pdf)

- [ ] AUDIO CODEC: Audio codec（芯唐电子 ISD9160 方案） 
- [ ] WIFI: MW31

supperthomas, 2021-12-06:
https://github.com/Notioni/OT_Demo

supperthomas, 2021-12-06:
https://github.com/robbie-cao/trumpet

supperthomas, 2021-12-07:
[image]

supperthomas, 2021-12-07:
http://mp.weixin.qq.com/s?__biz=MzA3MjMzNTc4NA==&mid=2649004067&idx=2&sn=9e67bfc070be73b48eee81e56fbec703&chksm=870fd0e4b07859f25520f005c636a36af5d563944bee4f3ebdc5105cad77a3f578af7966e76e&mpshare=1&scene=23&srcid=1207ofRm3TYsuB6F2gzuBRIe&sharer_sharetime=1638886299321&sharer_shareid=a0083c361d12ab1b3b3837a945e5ae42#rd

supperthomas, 2021-12-12:
玩转AliOS Things Developer Kit开发板博文汇总https://cloud.tencent.com/developer/article/1807468
https://blog.csdn.net/morixinguan/article/details/115154233
https://blog.csdn.net/morixinguan/article/details/102325789
https://www.pianshen.com/article/7270369866/

supperthomas, 2021-12-13:
超频120mhz

supperthomas, 2021-12-15:
https://github.com/bekencorp/bk7231_rtt_sdk

supperthomas, 2021-12-15:
3.	Alios things Wiki(这里资料很多)
https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Developer-Kit-Hardware-Guide

supperthomas, 2021-12-16:
https://www.wenjiangs.com/doc/link-develop-zp5fzh
发现这个网址，介绍大板传感器的

Jackistang, 2021-12-19:
Zephyr BSP: https://gitee.com/quincyzh/zephyr-for-developer-kit-stm32-l496

supperthomas, 2021-12-20:
> Zephyr BSP: https://gitee.com/quincyzh/zephyr-for-developer-kit-stm32-l496

欢迎PR到zephyr upstream上。