supperthomas/bluetoothlover_doc #272: [git]使用技巧总结

xupenghu, 2021-12-13:
## git error: src refspec dev does not match any
提交本地修改到云端时，报这个错误。是因为本地没有创建这个分支。
[image]
可以新建一个分支，然后再提交
```
$ git checkout -b wechat_private
$ git push origin wechat_private

```

supperthomas, 2021-12-13:
可以参考下文档：
https://supperthomas-wiki.readthedocs.io/en/latest/WIKI_FAQ/06_git_flow/git_flow.html

chenyingchun0312, 2021-12-13:
git rebase 目前了解到的两种用途：
1. 多个commit合并成一个 （这个比较有用）
2. 让git的提交记录变得简介（这个虽然骚气，但是感觉然并软？？）
[image]

hyhkjiy, 2021-12-14:
在终端使用ohmyzsh的git插件可以简化git大量操作：
- https://github.com/ohmyzsh/ohmyzsh/tree/master/plugins/git

hyhkjiy, 2021-12-14:
> 

Windows上面我也没试过，网上说可以配合MinGW的MSYS使用

hyhkjiy, 2021-12-14:
> > 在终端使用ohmyzsh的git插件可以简化git大量操作：
> > 
> > * https://github.com/ohmyzsh/ohmyzsh/tree/master/plugins/git
> 
> 我们对这个还是挺感兴趣的，下次开会可以分享下如何使用的吗？

Windows上面我也没试过，网上说可以配合MinGW的MSYS使用

chenyingchun0312, 2021-12-14:
> 在终端使用ohmyzsh的git插件可以简化git大量操作：
> - https://github.com/ohmyzsh/ohmyzsh/tree/master/plugins/git

这些，感觉都是一些别名，真正使用的话，其实太多命令的话，感觉别名更不容易记得，还是使用git GUI客户端更好一些吧，我目前使用git extension，感觉还挺好用的

hyhkjiy, 2021-12-15:
> > 在终端使用ohmyzsh的git插件可以简化git大量操作：
> > 
> > * https://github.com/ohmyzsh/ohmyzsh/tree/master/plugins/git
> 
> 这些，感觉都是一些别名，真正使用的话，其实太多命令的话，感觉别名更不容易记得，还是使用git GUI客户端更好一些吧，我目前使用git extension，感觉还挺好用的

在命令行的情况下可以用alias + grep快速查找需要的命令还是挺好用的，不过GUI客户端也方便，什么顺手就用什么吧 ^_^

supperthomas, 2022-05-20:
托管平台：
[image]

supperthomas, 2022-05-20:
https://mp.weixin.qq.com/s/QI33RSNd7RNExwLQJOLIng

supperthomas, 2022-08-23:
45 个 Git 操作场景，专治不会合代码

https://mp.weixin.qq.com/s/36sYeZYX9b-EqHnddZbLxw

supperthomas, 2023-01-08:
https://mp.weixin.qq.com/s/gooJKTZ6Sio7Q7Ls2kapxw
45个常见场景

supperthomas, 2023-01-13:
Git基本原理介绍

https://mp.weixin.qq.com/mp/publictag?action=get&tag_id=1993325228658262017&start=0&count=10&scene=1&sessionid=&enterid=0&from=groupmessage#wechat_redirect

supperthomas, 2023-03-14:
git pull --rebase
巧用git diff

supperthomas, 2023-04-04:
如何显示两个branch之间文件名差异

你可以使用以下 `git` 命令来获取 `branch1` 和 `branch2` 之间差异的文件名及路径：

```shell
git diff --name-only branch1 branch2
```

这个命令会输出 `branch1` 和 `branch2` 之间所有发生变更的文件的文件名及路径。如果你想要一个更详细的输出，包括文件差异的统计信息，可以使用 `--stat` 选项：

```shell
git diff --stat branch1 branch2
```

supperthomas, 2023-04-04:
diff_files=$(git diff --name-only $base_branch..$compare_branch)

    # 对每个差异的文件进行处理
    for file in $diff_files; do
        # 在这里添加你要执行的操作
        echo "Processing file: $file"
        # 假设你要执行的操作是将文件复制到另一个目录中
        cp $file /path/to/destination/
    done

hyhkjiy, 2023-04-04:
[实战：利用VS Code实现代码回滚到较远版本并保留新功能](https://jiy.notion.site/VS-Code-feb01202c918490f8ad055ebd0771849)

supperthomas, 2023-04-25:
git diff --name-only   可以显示当前差异文件。

supperthomas, 2024-09-16:
修改git默认编译器为vim

`git config --global core.editor "vim"`