supperthomas/bluetoothlover_doc #274: [RTT] PM学习

chenyingchun0312, 2021-12-15:
https://club.rt-thread.org/ask/article/2282.html

https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/pm/pm

chenyingchun0312, 2021-12-15:
pm_release
pm_release_all
pm_request
pm_module_release
pm_module_release_all
pm_module_request
pm_module_delay
pm_run
pm_dump

chenyingchun0312, 2021-12-15:
cubemx 路径： C:\Users\cheny\STM32Cube\Repository

chenyingchun0312, 2021-12-16:
接下来，先完成nordic pm驱动

chenyingchun0312, 2021-12-24:
https://github.com/supperthomas/supperthomas_doc/issues/232

chenyingchun0312, 2022-06-09:
[RT-Thread 电源管理组件-哔哩哔哩](https://b23.tv/KPn2NhK)

supperthomas, 2022-08-10:
[image]

supperthomas, 2023-04-14:
power down 之后，只能通过wakeup 唤醒，但是唤醒之后，会重新走reset流程。

supperthomas, 2023-04-14:
power down之后，可以通过RTC定时器来唤醒，power down之后，RTC还可以独立工作。

closed 2023-05-13 by chenyingchun0312