supperthomas/bluetoothlover_doc #283: 玩一下荔枝派zero

hyhkjiy, 2021-12-30:
好久之前买了一个荔枝派zero，因为烧录调试不方便等各种原因一直没玩起来，最近突然心血来潮画了个串口扩展板，准备玩玩。

扩展板链接：https://oshwhub.com/hyhkjiy/lichee_zero_uart_extend

相关资料
---

文档地址：https://licheezero.readthedocs.io/zh/latest/index.html

supperthomas, 2021-12-31:
@hyhkjiy 你有荔枝派吗？我有一块V3S

supperthomas, 2021-12-31:
@hyhkjiy 我也有个荔枝派，使用中欢迎多记录一些连接和资料到这个issue里。

hyhkjiy, 2021-12-31:
> @hyhkjiy 我也有个荔枝派，使用中欢迎多记录一些连接和资料到这个issue里。

扩展板到了
[image]

hyhkjiy, 2021-12-31:
元件也基本到齐，草率的是忘了家里没有10uF的电容，不过滤波电容嘛，滤不滤的估计区别不大。
[image]

hyhkjiy, 2022-01-01:
贴片完成，调试成功，记录一下步骤。

板子上锡
[image]

摆放元件
[image]

焊接+维修+洗板
[image]

插电之后显示设备
[image]

和另一个串口连在一起测试，互相都可以收到信息
[image]
[image]

焊接排母，引出脚针暂时没使用先不焊
[image]

插电，开机
[image]
[image]