supperthomas/bluetoothlover_doc #296: 【nimble】解决该issue

chenyingchun0312, 2022-01-22:
https://github.com/RT-Thread-packages/nimble/issues/13

supperthomas, 2022-01-22:
@xupenghu

chenyingchun0312, 2022-01-26:
gcc相关的汇编代码做了一些注释，以及替换后，这一块倒是没有报错了，但是ble uart的service编译报错

1. lastest版本中`ble_phy_rxpdu_copy`函数使用同名的v1.0.0版本中的函数替换
2. 注释掉其他报错的汇编代码

[image]

chenyingchun0312, 2022-01-26:
jrand48 报错问题
[image]

NIMBLE PR 链接： https://github.com/apache/mynewt-nimble/pull/883  

jrand48是Linux的系统调用，并不是标准的C库接口，暂时使用rand替换该函数

jrand48 man手册： https://linux.die.net/man/3/jrand48

[image]

chenyingchun0312, 2022-01-26:
还有好几个错误：

[image]

chenyingchun0312, 2022-01-26:
@supperthomas 涛哥，最后的报错问题，我看了下，是ble uart的service报错了，在gcc 平台也测试出错了，这个之前在哪个平台编译通过了么？

[image]

supperthomas, 2022-01-27:
@chenyingchun0312 可以换个sample试下。比如peripheral。

supperthomas, 2022-01-27:
`https://github.com/RT-Thread-packages/nimble/issues/11`

supperthomas, 2022-01-27:
@chenyingchun0312 我刚才试了下，很多sample都可以用的哦。比如peripheral_heart_rate。用scons是可以编译通过的，你可以试试其他sample.

chenyingchun0312, 2022-02-14:
@supperthomas 涛哥，我看了一下，貌似所有的sample都会编译有问题，不管gcc还是MDK平台，因为编译报错的文件是`nimble/host/services/bleuart/src/bleuart.c`，而这个文件属于Host层的文件，和sample不相关，并且都会参与编译，不知道之前真的编译过了？

感觉怪怪的，目前对nimble结构还没仔细研究，有待解决该问题

[image]
[image]

supperthomas, 2022-02-14:
@chenyingchun0312 直接PR吧。

chenyingchun0312, 2022-02-23:
涛哥，这个nimble的问题，我应该只能协助调查原因，真正解决，我觉得，还需要换一个对nimble比较熟悉的ownner，因为我的解决比较粗暴。

supperthomas, 2022-02-23:
@chenyingchun0312 可以讲讲你的理解。

supperthomas, 2022-02-23:
@chenyingchun0312 建议，改动不大，先PR，我可以来讲讲。

chenyingchun0312, 2022-02-25:
测试不充分，我只测试了bleuart的demo

https://github.com/supperthomas/supperthomas_doc/files/8061766/0001-.txt

[image]

supperthomas, 2022-02-25:
@chenyingchun0312 PR吧。