supperthomas/bluetoothlover_doc #303: Rust 学习资料

Jackistang, 2022-03-12:
书籍：
- [Rust程序语言设计 - 中文版](https://rustwiki.org/zh-CN/book/title-page.html)
- [通过例子学 Rust 中文版](https://rustwiki.org/zh-CN/rust-by-example/)
- [Rust 命令行应用开发教程](https://rust-cli.github.io/book/index.html)

入门学习项目：
- 利用 Rust 构建一个简单的 Shell：https://www.joshmcguigan.com/blog/build-your-own-shell-rust/
- 通过写一个链表学习 Rust：https://github.com/rust-unofficial/too-many-lists
- 利用 Rust 学习解析器组合器：https://bodil.lol/parser-combinators/

Rust 开源项目集合：
- awesome-rust: https://github.com/rust-unofficial/awesome-rust
- awesome-embedded-rust: https://github.com/rust-embedded/awesome-embedded-rust#raspberry-pi

其他：
- Rust 软件包集合：https://crates.io/
- Rust 中文技术论坛：https://rustcc.cn/
- Rust 标准库：https://rustwiki.org/zh-CN/std/

supperthomas, 2022-03-12:
安装rust:
https://www.rust-lang.org/zh-CN/tools/install

`curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh`

supperthomas, 2022-04-04:
【Rust嵌入式开发入门(01)--系列教程概览-哔哩哔哩】 https://b23.tv/dNu5wj9

supperthomas, 2022-04-17:
rust 圣经
https://course.rs/about-book.html