supperthomas/bluetoothlover_doc #318: RISC-V 文档完善

chenyingchun0312, 2022-04-19:
RTT官网RISC-V文档完善

chenyingchun0312, 2022-04-19:
官网连接：

https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/application-note/porting/risc-v/port-risc-v?id=risc-v-%e7%ae%80%e4%bb%8b

chenyingchun0312, 2022-06-17:
https://github.com/chenyingchun0312/rt-thread/tree/riscv/libcpu/risc-v

注释主要在这里，等把手头GUI程序搞定,就把这个文档完善了,感觉应该对别人有帮助

[image]

chenyingchun0312, 2022-06-17:
之前整理的issue   https://github.com/supperthomas/supperthomas_doc/issues/200

chenyingchun0312, 2022-06-19:
RISC-V 指令结构

[image]

chenyingchun0312, 2022-06-26:
【RISC-V处理器嵌入式开发课程-哔哩哔哩】 https://b23.tv/puaMjLt

[image]

supperthomas, 2022-06-30:
【RISC-V：intel 、ARM ，你 怎 么 睡 得 着 觉 ？-哔哩哔哩】 https://b23.tv/gUBXb7N