supperthomas/bluetoothlover_doc #323: 【log】RT-THREAD log系统介绍

supperthomas, 2022-05-10:
1 ulog
https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/ulog/ulog?id=ulog-%e7%ae%80%e4%bb%8b

supperthomas, 2022-05-10:
```
#define LOG_TAG     "example"    
#define LOG_LVL     LOG_LVL_DBG  
#include <ulog.h>               
```

supperthomas, 2022-05-10:
目前 ulog 已集成控制台后端，即传统的输出 rt_kprintf 打印日志的设备。ulog 还支持 Flash 后端，与 EasyFlash 无缝结合，详见其软件包（[点击查看](https://github.com/armink-rtt-pkgs/ulog_easyflash_be)）。后期 ulog 还会增加文件后端、网络后端等后端的实现。当然，如果有特殊需求，用户也可以自己来实现后端。

supperthomas, 2022-05-10:
[ulog_file](https://github.com/RT-Thread-packages/ulog_file.git)ulog_file: 基于文件系统的 ulog 后端插件。  这个比较方便只需要调用ulog_file_backend_init 这个函数就可以了。
ulog的东西会写到文件系统里面。

```
//#include <dfs_posix.h>
```
最新的这个需要注释掉。

xupenghu, 2022-05-29:
有没有一种可能 让ulog输出的日志可以使用脚本分析日志内容，从而自动化分析日志

supperthomas, 2022-05-29:
> 有没有一种可能 让ulog输出的日志可以使用脚本分析日志内容，从而自动化分析日志

现在ulog已经可以生成txt或者其他形式的log了。拿出这个log之后，可以做个小脚本，自动化分析，比如根据模块来过滤

supperthomas, 2022-05-29:
RT-Thread Studio ulog日志
https://blog.csdn.net/weixin_37875741/article/details/109830100

chenyingchun0312, 2022-05-31:
> 有没有一种可能 让ulog输出的日志可以使用脚本分析日志内容，从而自动化分析日志

xshell 等串口软件，支持LOG实时输出到文件，并且记录时间戳，可以通过脚本解析

closed 2022-11-10 by supperthomas

supperthomas, 2022-11-21:
- [ ] [EasyLogger](https://github.com/armink-rtt-pkgs/EasyLogger)	armink	MIT	一款超轻量级(ROM<1.6K, RAM<0.3k)、高性能的 C/C++ 日志库

- [ ] [logmgr](https://github.com/RT-Thread-packages/logmgr.git)	ChenYong	Apache-2.0	logmgr: 日志管理系统功能支持
- [ ] [lwlog](https://github.com/wuhanstudio/lwlog)	wuhanstudio	MIT	lwlog:单文件日志打印库
- [ ] [ulog_easyflash](https://github.com/armink-rtt-pkgs/ulog_easyflash)	armink	MIT	基于 EasyFlash 的 ulog 插件
- [ ] [ulog_file](https://github.com/RT-Thread-packages/ulog_file.git)	ChenYong	Apache-2.0	ulog_file: 基于文件系统的 ulog 后端插件。