supperthomas/bluetoothlover_doc #361: zepgyr轮询结构调整

bobwenstudy, 2022-08-06:
目前正在处理net_buf，调整为轻量级的buf管理结构。

supperthomas, 2022-08-06:
scons

bobwenstudy, 2022-08-20:
1. net_buf重构，调整为fifo pool，以便于tx/rx支持多线程处理
2. 调整文件目录 修改接收行为和bt行为共线程 后续处理rx buffer溢出的问题
3. 日志机制优化，开启printf会delay，默认关闭printf行为；加入critical打印等级，满足界面打印日志需要，printf打印太多系统会卡住
4. central+peripheral，demo全部完成。
5. 修正master发起加密的行为。
6. 修正收到adv太多太快，导致rx buffer溢出。
7. id/smp/hci/key，相关代码合入zephyr最新代码。
8. 调整代码format，zephyr的蓝牙代码并没有用项目中clang-format的.code_format，为方便比较代码，后续两边都方便比较差异。

bobwenstudy, 2022-09-03:
1. 同步zephyr最新代码，后续陆续开始测试
2. 为确保和zephyr库的代码尽量一致，将原本改掉的接口变回来，保留接口不变，实现方式适当调整。timeout\k_fifo等。
3. 加入吞吐测试的case，多次连接断开工作正常。
4. 加入接收buffer溢出处理
5. 修正extern "c"的影响
6. 所有头文件都需要用#ifndef包起来
7. 梳理头文件，确保bt_config.h干净，进一步梳理文件分布和头文件引用逻辑
8. 加入Kconfig框架
9. 调整编译行为，为了方便后续大家使用，暂时去除scons逻辑，使用makefile编译模式
10. 所有文件加入bt_config.h引用，方便后面移植。
11. 清空所有警告
12. 调整日志管理行为 关闭gatt service开关行为