supperthomas/bluetoothlover_doc #398: 关于急性痛风性关节炎的一些介绍

xupenghu, 2022-09-19:
## 背景

        中秋回去因一些不可靠因素，喝了一些白酒，第二天早上右脚大拇指关节开始肿胀，以为是前一晚酒醉踢到了什么地方，没太关注。当天去西安后，疼痛和肿胀继续加重，遂去医院挂了急诊，拍了片子。医生解释一切正常，没有骨折。这个时候，还没往痛风方向想（之前也没得过，周围人也没得过，对这个不了解），买了一些云南白药、跌打丸之类的药，用了两天不见任何好转。期间自行百度了相关知识，察觉是痛风，刚好同事有测尿酸的仪器，遂测了一下尿酸440（正常值149~416），然后去诊所买了相应止疼药和降尿酸的药，当天疼疼痛缓解。周六去医院验血复查尿酸262，属于正常值，周天疼痛、肿胀缓解。

## 急性痛风性关节炎的一些症状

         急性[关节炎](https://baike.baidu.com/item/%E5%85%B3%E8%8A%82%E7%82%8E/792952?fromModule=lemma_inlink)期 多在夜间突然发病，受累关节剧痛，首发关节常累及第一跖趾关节，其次为踝、膝等。关节红、肿、热和压痛，全身无力、发热、[头痛](https://baike.baidu.com/item/%E5%A4%B4%E7%97%9B/340011?fromModule=lemma_inlink)等。可持续3～11天。饮酒、暴食、过劳、着凉、手术刺激、精神紧张均可成为发作诱因。

       如果出现脚大拇指关节突然疼痛、肿胀，又没有明显外伤，那基本可以判定是急性痛风性关节炎，赶紧对症用药治疗，痛起来是真要命。

xupenghu, 2022-09-19:
痛风常发作于这个部位
[image]

xupenghu, 2022-09-19:
## 参考文章
- http://wjw.beijing.gov.cn/bmfw_20143/jkzs/jbzs/201912/t20191217_1250025.html
- https://baike.baidu.com/item/%E7%97%9B%E9%A3%8E%E6%80%A7%E5%85%B3%E8%8A%82%E7%82%8E/5085951
-

supperthomas, 2022-09-19:
多喝水可以增加尿量，减少尿酸结石形成机会，缩短痛风发作的持续时间。建议高尿酸血症者，每日尿量维持在2000-3000ml，但应避免饮用可乐、橙汁、苹果汁等含果糖饮料或含糖软饮料、果汁和浓汤，可食用含果糖少的水果（樱桃、草莓、菠萝、西瓜、桃子等），可以饮用淡茶或不加糖的咖啡。

chenyingchun0312, 2022-09-19:
还是老话说得好呀，得多喝水呢

supperthomas, 2022-09-19:
```
第一类食物是含嘌呤最高的，之后依次降低。
第一类含嘌呤高的食物(每100g食物含嘌呤100～1000mg)
肝、肾、胰、心、脑、肉馅、肉汁、肉汤、鲭鱼、风尾鱼、沙丁鱼、鱼卵、小虾、淡菜、鹅、斑鸡、石鸡、大豆制品、酵母、香菇、紫菜。

第二类含嘌呤中等的食物(每100g食物含嘌呤75～100mg)
1.鱼类：鲤鱼、鳕鱼、大比目鱼、鲈鱼、梭鱼、贝壳类、鳗鱼及鳝鱼。
2.肉食：熏火腿、猪肉、牛肉、牛舌、小牛肉、兔高嘌呤肉、鹿肉。
3.禽类：鸭、鸽子、鹌鹑、野鸡、火鸡。

第三类含嘌呤较少的食品(每100g食物含嘌呤<75mg)
1.鱼蟹类：青鱼、鲱鱼、鲑鱼、鲥鱼、金枪鱼、白鱼、龙虾、蟹、牡蛎。
2.肉食：火腿、羊肉、牛肉汤、鸡、熏肉。
3.麦麸:麦片、面包、粗粮。
4.蔬菜：芦笋、菠菜、蘑菇。

第四类含嘌呤很少的食物
1.粮食：大米、小麦、小米、荠麦、玉米面、精白粉、富强粉、通心粉、面条、面包、馒头、苏打饼干、黄油小点心。
2.蔬菜：白菜、卷心菜、胡萝卜、芹菜、黄瓜、茄子、甘蓝、芜青甘蓝、甘蓝菜、莴笋、刀豆、南瓜、倭瓜、西葫芦、蕃茄、山芋、土豆、泡菜、咸菜。
3.水果：各种水果。
4.蛋、乳类：鲜奶、炼乳、奶酪、酸奶、麦乳精。
5.饮料：汽水、茶、咖啡、可可、巧克。
其他食物：各种油脂、花生酱、洋菜冻、果酱、干果等。[2]

无或低嘌呤

无嘌呤或含低嘌呤食物，
如精粉、大米、苏打饼干、馒头、面包、奶类及奶制品、蛋类、各类油脂、水果、干果、糖及糖果；
除菜花、菠菜等少数蔬菜外，大部分蔬菜，如胡萝卜、芹菜、卷心菜、黄瓜、茄子、西红柿、西葫芦、土豆含嘌呤都很少
```

xupenghu, 2022-09-19:
> ## 背景 [...]

## 不要喝酒！！！！

supperthomas, 2022-09-20:
> ## 不要喝酒！！！！

有没有吃螃蟹？

closed 2022-11-10 by supperthomas