supperthomas/bluetoothlover_doc #425: chatgpt人工智能聊天

supperthomas, 2022-12-11:
【淘宝】https://m.tb.cn/h.URj3VtD?tk=Ajvbd34d9q6 CZ0001 「chatgpt注册账号户直接登陆攻略教程人工智能中文对话openai」
点击链接直接打开 或者 淘宝搜索直接打开

supperthomas, 2022-12-12:
https://gpt.chatapi.art

supperthomas, 2022-12-12:
https://chat.openai.com/

supperthomas, 2022-12-28:
[image]

CSDN 这个工具可以免费使用CHATGPT

supperthomas, 2023-01-04:
应用场景1：
写正则表达式

supperthomas, 2023-01-06:
Wow，基于ChatGpt的搜索上线了！www.you.com

supperthomas, 2023-02-15:
https://aigcfun.com
https://xc.com
http://chat.h2ai.cn/home

supperthomas, 2023-02-26:
添加新成员chatgpt

supperthomas, 2023-03-06:
应用场景2：
写python脚本

supperthomas, 2023-03-06:
https://blog.csdn.net/qq_33866817/article/details/129122178

supperthomas, 2023-03-06:
【新版Bing——科研人的终极解决方案-哔哩哔哩】 https://b23.tv/lTTjiR4

supperthomas, 2023-03-07:
[image]

supperthomas, 2023-03-08:
vscode 插件不是基于chat的，不是特别好用。

supperthomas, 2023-03-08:
【集大成者！ChatGPT百宝箱，内置多种功能，所见即所得！-哔哩哔哩】 https://b23.tv/YrUnwd1

supperthomas, 2023-03-08:
Notion AI

supperthomas, 2023-03-08:
copyleaks

supperthomas, 2023-03-08:
AI 绘画， STABLE DIFFUSION

supperthomas, 2023-03-09:
[sms-activate](https://sms-activate.org/)

supperthomas, 2023-03-09:
https://github.com/f/awesome-chatgpt-prompts

supperthomas, 2023-03-10:
【【爆炸神器copilot】写点注释，代码自动生成-哔哩哔哩】 https://b23.tv/MquodoBl

supperthomas, 2023-03-10:
https://chat35.com/i/TaLhWZ

supperthomas, 2023-03-10:
https://www.mojidoc.com/05z7y-dd5pa7hu3zfmhnbngoeztyqcnq-00b/?utm_medium=sns&utm_source=twiitter&utm_content=0215

汇总

chenyingchun0312, 2023-03-11:
指导chatGPT的命令列表：
帮我总结这些commit message的信息，生成一个可读性强的、分类好的、中文的发布公告

chenyingchun0312, 2023-03-11:
从现在起，当你想发送一张照片时，请使用Markdown，并且不要有反斜线，不要用代码	
块。使用Unsplash APIhttps://source.unsplash.com/960x640/?<英语关键词>)如果你明白了，请回复“明白”

supperthomas, 2023-03-12:
https://github.com/PlexPt/awesome-chatgpt-prompts-zh

supperthomas, 2023-03-12:
https://github.com/prompt-engineering/understand-prompt

supperthomas, 2023-03-12:
【整合chatGPT的新必应（NewBing chat）简直就是科研神器！-哔哩哔哩】 https://b23.tv/0VOjVBj

supperthomas, 2023-03-12:
edge dev    https://www.microsoftedgeinsider.com/zh-cn/download/dev
skype   https://www.skype.com/zh-Hans/get-skype/

supperthomas, 2023-03-13:
【过把AI驯化师的瘾，用ChatGPT API做嵌入式串口协议对接！-哔哩哔哩】 https://b23.tv/5x11imf

supperthomas, 2023-03-14:
www.chatpdf.com

chenyingchun0312, 2023-03-14:
edge dev可以读PDF

[image]

supperthomas, 2023-03-14:
http://ai.okmiku.com/chat

supperthomas, 2023-03-16:
https://github.com/zhayujie/chatgpt-on-wechat

chatgpt聊天机器人

supperthomas, 2023-03-16:
https://cloud.baidu.com/survey_summit/wenxin.html 
百度文心一言申请

bobwenstudy, 2023-03-17:
https://promptszone.com/chat?hl=zh-CN
大家可以试试这个，不翻墙也可以用，试了下，随便填一个邮箱就验证下就可以用了
体验和chatgpt一致
[image]

supperthomas, 2023-03-17:
https://chatgpt.denohub.com/
chatgpt 从入门到精通。这个写的不错，都是最直接的方式使用chatgpt。

supperthomas, 2023-03-18:
https://cloud.baidu.com/survey_summit/wenxin.html 文星一眼可以申请了

bobwenstudy, 2023-03-20:
https://freegpt.one/，（需科学上网），可直接使用
[image]

supperthomas, 2023-03-20:
https://www.cursor.so/  可以辅助写代码 还可以直接对话 速度挺快的

supperthomas, 2023-03-29:
https://codeium.com/

copikot替代品

supperthomas, 2023-04-02:
[image]

supperthomas, 2023-04-04:
https://mp.weixin.qq.com/s/qoE7vTvjdJTCDYvA2FO9-g

supperthomas, 2023-04-04:
AI工具合集
【上图中】网站汇总 #AI
📝AI文本
ChatGPT：https://chat.openai.com/
NotionAI：https://www.notion.so/product/ai
A.I. Data Sidekick：AI工具编写 SQL、文档等的速度提高10倍
https://www.airops.com/
Writesonic：人工智能写作辅助工具
https://writesonic.com/
copy.ai：使用 AI 编写更好的营销文案和内容
https://www.copy.ai/
Character.AI：AI人工交互
https://beta.character.ai/
Fireflies：该工具可插入 Zoom、Teams 或 Webex 等流行的视频会议工具，并自动执行做笔记和创建转录的过程
https://fireflies.ai/
Jasper： AI文案写作工具
https://www.jasper.ai/
Outplay：https://outplayhq.com/
CoWriter：AI辅助写作
https://cowriter.org/login

🎨AI绘画
Midjourney：AI绘画神器
网站：https://www.midjourney.com/
教程：https://www.uisdc.com/midjourney
PhotoRoom：擦除任何背景、对象
https://www.photoroom.com/
造梦师：只需一句话，让你的文字变成画作
https://printidea.art/
ARC Lab：一款提供照片修复、抠图、画质增强的在线工具
https://arc.tencent.com/zh/ai-demos/faceRestoration
Artbreeder：人工智能合成创意工具
https://www.artbreeder.com/
Stockimg AI：生成各种各样的设计元素,包括logo、插画、图片壁纸等
https://stockimg.ai/
niji·journey：二次元ai绘画
https://nijijourney.com/zh/
Getimg.ai：关键词生成图片的AI工具
https://getimg.ai/
Dreamlike.art：AI图像生成
https://dreamlike.art/
文心一格 飞桨：AI艺术和创意辅助平台
https://yige.baidu.com/
Phygital+：AI图像生成
https://phygital.plus/
Beautiful.ai：AI生成PPT
https://www.beautiful.ai/

🎶AI音频
Brain.fm：专注、放松、冥想和睡眠，聆听为您的大脑量身打造的音乐
https://www.brain.fm/
Soundraw：人工智障生成音乐
https://soundraw.io/
Endel：个性化背景音，帮助您集中注意力、放松和睡眠
https://endel.io/
Riffusion：实时音乐和音频生成库
https://www.riffusion.com/
https://github.com/riffusion/riffusion
Papercup：人工智障配音和视频翻译软件
https://www.papercup.com/
LALAL.AI：从任何音频和视频中提取人声、伴奏和各种乐器
https://www.lalal.ai/
Murf：使用多功能AI语音生成器从文本到语音
https://murf.ai/
PolyAI：语音助手
https://poly.ai/
Voicemod：语音实时变声器
https://www.voicemod.net/zh/
Boomy：人工智障生成音乐
https://boomy.com/
Mubert：人工智障生成音乐
https://mubert.com/

🎞AI视频
Runway： AI 魔法工具
https://runwayml.com/
Cascadeur：人工智障辅助关键帧动画软件
https://cascadeur.com/
Synthesia：人工智障视频生成
https://www.synthesia.io/
Pollinations：文本转视频、图片
https://pollinations.ai/
Zubtitle：为视频添加字幕和在线编辑视频
https://zubtitle.com/
Munch：人工智障提取视频中的最引人入胜、最流行和最有影响力的片段
https://www.getmunc

supperthomas, 2023-04-04:
https://app.rytr.me/create

supperthomas, 2023-04-06:
https://github.com/weekend-project-space/chatgpt-sites

supperthomas, 2023-04-07:
综合性网站，挺好的
https://cc.ai55.cc/

supperthomas, 2023-05-22:
https://github.com/chatanywhere/GPT_API_free

supperthomas, 2024-02-07:
https://chat.aimakex.com/

supperthomas, 2024-08-03:
kimi.ai

supperthomas, 2024-08-03:
sider edge

supperthomas, 2024-08-03:
一键生成PPT

supperthomas, 2024-08-03:
qanything.ai

supperthomas, 2024-08-03:
liblib.art

supperthomas, 2024-08-07:
ai-bot.cn

supperthomas, 2024-08-07:
提示词
prompthero.com

supperthomas, 2024-08-08:
vip8.1ai.ink    画图和海报，各种引擎

supperthomas, 2024-08-20:
https://www.onlyoffice.com/zh/download-desktop.aspx   office

supperthomas, 2025-02-19:
```
name: Code Review

permissions:
  contents: read
  pull-requests: write

on:
  pull_request:
    types:
      - review_requested

jobs:
  review_full:
    if: github.event.action == 'review_requested' && github.event.requested_reviewer.login == 'Nordicbuilder'
    runs-on: ubuntu-latest
    steps:
      - name: Full PR Diff Review
        uses: NordicBuilder/ChatGPT-CodeReview@takeAllCommits
        env:
          GITHUB_TOKEN: ${{ secrets.NORDICBUILDER_GPT }}
          OPENAI_API_KEY: ${{ secrets.OPENAI_API_KEY }}
          IGNORE_PATTERNS: /placeholder
          INCLUDE_PATTERNS: "*.*"
```

supperthomas, 2025-02-19:
https://github.com/anc95/ChatGPT-CodeReview/blob/main/README.zh-CN.md