supperthomas/bluetoothlover_doc #440: 开源之夏参考文章和链接

supperthomas, 2023-03-25:
查看下面视频，可以简单快速过一遍，主要目的：理解什么是开源精神。

- [开源软件供应链 2020峰会 分会场-操作系统&编译器](https://www.bilibili.com/video/BV17Z4y1G7TX)

- [开源软件供应链 2020 峰会-优秀学生分享](https://www.bilibili.com/video/BV1ky4y1q7n2)

- [开源软件供应链 2020 峰会-主会场](https://www.bilibili.com/video/BV1yi4y1V7rd)

- [这十个思维方式如何改变我的人生（2020）](https://www.bilibili.com/video/BV1A5411E7zG)

supperthomas, 2023-03-25:
本文主要放一些暑期开源相关资料

- [summer2020所有项目](https://isrc.iscas.ac.cn/summer2020/#/announcement)

- [summer2020视频资料录制](https://isrc.iscas.ac.cn/summer2020/#/summitmeeting)

- [暑期 2021 名单公示：681 位学生中选](https://mp.weixin.qq.com/s/DhGW24v04goyh3htoRTdag)

- [summer2020峰会论坛](https://wx.vzan.com/live/tvchat-425619793#/)  必看

- [官方仓库](https://github.com/isrc-cas)

为期 3 个月，项目开发全面开始

图片
7 月 1 日-9 月 30 日为项目开发工作周期，在此期间，中选学生将在导师的指导下独立完成相应的开发任务。
项目开发分为两个阶段，对应两个重要里程碑：中期考核、结项考核。
阶段一：7 月 1 日-8 月 15 日
阶段二：8 月 16 日-9 月 30 日
注：只有通过中期考核的同学才可以继续第二阶段的项目开发。

开发过程中，根据社区要求产出成果可以存放在社区指定平台的公开仓库，也可以选择存放在组委会提供的仓库；需要注意的是，学生需要在里程碑节点之前向组委会平台仓库提交完整项目代码，并将相应的中期和结项报告提交至系统平台，接受导师的评审。

supperthomas, 2023-03-25:
《大教堂和集市》建议严读
https://www.ruanyifeng.com/blog/2008/02/notes_on_the_cathedral_and_the_bazaar.html

supperthomas, 2023-03-28:
TODO LIST:
1. 完善下群里的doc题目。
2.

bobwenstudy, 2023-03-29:
题目3：蓝牙方向
导师姓名： 李涛、文波
导师邮箱： wenbo13579@163.com
项目难度：基础
大体方向：stm32+zephyr_polling+uart_dongle
目的:
进一步验证zephyr_polling项目的适配性，确保可以在不同平台上接入uart蓝牙芯片就可以正常使用，方便RT-Thread在蓝牙使用上不在局限于具体蓝牙芯片平台（如ESP32，nrf52840，stm32等），后续有需要的任何人在单片机系统中，可以快速接入蓝牙业务，
产出标准：
1.定期的输出学习文档
2.定期PR贡献代码
3.选定uart蓝牙模块（暂定nrf52840，stm32-blueNRG1,2，也可自己选择蓝牙HCI模块），验证模块HCI通信功能正常。
4.选定单片机（如ART-PI、stm32L496ZG等）系统，stm32单片机芯片正常运行RT-Thread功能，porting zephyr_polling协议栈，并和外接模块正常运行蓝牙功能业务。
5.在bsp上完成指定单片机移植验证。
6.在RT-Thread Studio上完成指定单片机demo开发和上传。
7.最后完成压力测试的验证，并通过压力测试来验证蓝牙的稳定性
技术要求：
1.愿意学习蓝牙相关的基础知识
2.愿意学习rtthread相关贡献
3.具有开源奉献精神
4.有一定的学习时间

supperthomas, 2023-04-21:
https://github.com/RT-Thread/rt-thread/tree/master/examples/utest/testcases/drivers