supperthomas/bluetoothlover_doc #467: [gitlab][CI] 添加CI脚本，编译MDK工程

chenyingchun0312, 2023-05-08:
【GitLab的cicd自动发布构建流程-哔哩哔哩】 https://b23.tv/1h0QsM4

chenyingchun0312, 2023-05-08:
before_script:
  - ​ec​ho "before_script!"
  
variables:
   DOMAIN: example.com
   
stages:
   - build
   - test
   - codescan
   - deploy
   
build:
  before_script:
  - echo "before_script in job"
  stage: build
  script:
    - echo "mvn clean"
    - echo "mvn install"
    - echo "$DOMAIN"
  after_script:
    - echo "after script in buildjob"
    
unittest:
    stage: test
    script:
      - echo "run test"
      
deploy:
  stage: deploy
  script:
    - echo "codescan"
    - sleep 2;
    
codescan:
  stage: codescan
  script:
    - echo "codescan"
    - sleep 5;
    
after_script:
  - echo "after_script!!!!"
  - ech​

chenyingchun0312, 2023-05-08:
wget https://mirrors.tuna.tsinghua.edu.cn/​gitla​b-ce/yum/el7/gitlab-ce-12.9.0-ce.0.el7.x86_64.rpm​

chenyingchun0312, 2023-05-08:
docker run -d \
--publish 443:443 --publish 80:80 --publish 1001:22 \
--name gitlab \
--volume gitlab_config:/etc/gitlab \
--volume gitlab_logs:/val/log/gitlab \
--volume gitlab_data:/val/opt/gitlab \
--shm-size 4gb \
gitlab/gitlab-​c​e:latest

-------------docker exec -it gitlab grep 'Password:' /etc/gitlab/initial_root_password​

chenyingchun0312, 2023-05-08:
【gitlab-cicd-保姆式讲解从零到一搭建部署流程-哔哩哔哩】 https://b23.tv/9PwsjvW

supperthomas, 2023-05-09:
gitlab的CI yml语法是不是和github是一样的？

chenyingchun0312, 2023-05-11:
【DevOps案例：Gitlab CI/CD 自动化部署项目-哔哩哔哩】 https://b23.tv/BMooAmL

chenyingchun0312, 2023-05-11:
> gitlab的CI yml语法是不是和github是一样的？

涛哥，看上去，不太一样

supperthomas, 2023-05-11:
https://docs.github.com/zh/actions/migrating-to-github-actions/manual-migrations/migrating-from-gitlab-cicd-to-github-actions

从 GitLab CI/CD 迁移到 GitHub Actions

chenyingchun0312, 2023-05-11:
windows环境，runner配置时，需要配置为`poweshell`

chenyingchun0312, 2023-05-11:
CI脚本，调用windows中的一些脚本，工具，比如python解析器时，需要将python解释器的路径放到环境变量的PATH中，而不能在其他环境变量中，比如PYTHON_PATH中

supperthomas, 2023-07-10:
https://mp.weixin.qq.com/s/Kr5i3Vg0zwPPMD2Bbmlt_Q

supperthomas, 2024-01-04:
继电器控制器
https://new.m.taobao.com/detail.htm?id=666971997975&ut_sk=1.ZDtChbpDhTgDAFjI3T8p6HYJ_21646297_1703673976159.TaoPassword-WeiXin.1&un=a515d723afacc92d46e6040d6f65052e&share_crt_v=1&un_site=0&spm=a2159r.13376460.0.0&sp_abtk=gray_1_code_simpleAndroid2&tbSocialPopKey=shareItem&sp_tk=cHNJS1dUU1ZybGs%3D&cpp=1&shareurl=true&short_name=h.5MDiVxU&bxsign=scdDh8uexaIKFBnfzgHcrhipjQPWAd7dHF5PX0UO4CsASmV3ZECcnt9qXTUbPyQ8cTTx_kJFkXkgaLqFbvdYN2WPXfMma2i9a601-tpORIcKl8s_n7fDrBr2l7kaLawtH6j&tk=psIKWTSVrlk&app=chrome